RMerl/asuswrt-merlin.ng #329: N66U and AC66U supports 382, where is merlin support?

caniko2, 2019-06-29:


RMerl, 2019-06-30:
These two models are no longer supported by Asuswrt-Merlin.

closed 2019-06-30 by RMerl

mpratt14, 2019-06-30:
@RMerl If what he said is true, it's kind of a good point. What exactly is the reasoning its not supported? Did you talk about it anywhere else? Maybe put the reason on your website so you don't get bugged about it anymore?

asaddon, 2019-06-30:
@mpratt14 Merlin no longer supports older MIPS platform and that's the reason these two routers are no longer supported. 

Full announcement here:
https://www.snbforums.com/threads/end-of-year-2017-development-update.42965/

mpratt14, 2019-06-30:
Thanks, @asaddon that's a very good thread to have, very detailed.

It should also answer all your questions @caniko2 . In the thread, Merlin mentioned "John's Fork" as still supporting those models, although that was more than a year ago. Maybe browse through the forks on the legacy repository and you may find something current

caniko2, 2019-07-02:
Would you suggest using the commercial versions instead? I don't need the bells and whistles of Asuswrt Merlin, I just want the stability.