RMerl/asuswrt-merlin.ng #332: empty “Client status” & “View List”

KayLeung, 2019-07-13:
“Client status” & “View List” are empty. It's broken since 384.11 (8-May-2019).

RMerl, 2019-07-13:
Use a supported models.  Modded TM-AC1900 and illegal firmware variants are not supported, and lead to that kind of issues among other things.

closed 2019-07-13 by RMerl

KayLeung, 2019-07-13:
Thanks.

BTW,
the update_clients.asp response the correct data but just haven’t inserted. It seems like the web UI issues. For anyone hit the same issue and want to check the list, you can copy the response into Console and check.

eruisi, 2020-07-20:
Before we get fix in the firmware, is there a way to get client list from command line? or some special URL?

KayLeung, 2020-07-20:
@eruisi 

The data stored in `originData`. You can access it from browser's Console

https://github.com/RMerl/asuswrt-merlin.ng/blob/129bcaf0486caef896bc038f144e2bf1856c82b1/release/src/router/www/client_function.js#L96