RMerl/asuswrt-merlin.ng #333: LAN - DHCP Server page doesn't escape single quotes

dhrubins, 2019-07-13:
When a device on the network gets named with a single quote, like "Dan's iPhone", the code [here](https://github.com/RMerl/asuswrt-merlin.ng/blob/5e862e3aed685260b4fd66e744cf823b52f485da/release/src/router/www/Advanced_DHCP_Content.asp#L97) chokes because it doesn't escape single quotes. This is how the code looks when rendered: 
`var dhcp_staticlists = '#6000:00:00:00:00:00&#62192.168.1.100&#62Dan's iPhone';`
Unfortunately, this prevents the rest of the page from loading.

crkinard, 2019-07-13:
Not that surprising considering ' is not a valid charter to be used in a hostname.

dhrubins, 2019-07-13:
Closing issue - my mistake. 
My router is running 380.xx, which has this problem [here](https://github.com/RMerl/asuswrt-merlin/blob/9f14d213d07fa36da459424a699bfe85f15b2286/release/src/router/www/Advanced_DHCP_Content.asp#L84), because this bit of code was still single-quoted in the old codebase. It's now double-quoted in the new codebase, so I'll submit a PR on the old repo. Unfortunately, I'm on a RT-AC66 so I'm stuck with 380.xx.

closed 2019-07-13 by dhrubins