RMerl/asuswrt-merlin.ng #336: Display hostnames in traffic monitor

Eldenroot, 2019-07-20:
[image]

As title says, would be nice to add hostnames to the table to quickly know the name of device.

Moreover scale option - is possible to add a TB?

RMerl, 2019-07-24:
Click on "Display Advanced Options", there's an option there to display hostnames.

Eldenroot, 2019-07-24:
I wil check that.

It will display IP and hostname together? I will check when I return from work.

Eldenroot, 2019-07-24:
@RMerl I annot find it, where is the setting located? Thank you! Screenshot please :)

RMerl, 2019-07-25:
[image]

However I can't really test that particular feature any longer since it's not compatible with the 4.1 kernel used by newer routers.

closed 2019-07-25 by Eldenroot