RMerl/asuswrt-merlin.ng #337: Caching issue regarding dns over tls

doracat55555, 2019-07-23:
When enabled dns over tls, the caching on dnsmasq and ttl is not honored, and IP not refreshed even after router is rebooted.

Tried:
1) set dnsmasq cache size to 0
2) local ttl to 0
3) restart dnsmasq and no luck
4) didn’t find anything related in Stubby to configure caching( i think is not supported yet)

Repro steps:
1) enable dns over tls (my test case use cleanbrowsing dns)
2) in dnsfilter, set to router
3) dhcp dns set to router IP
4) toggle block/allow in cleanbrowsing or opendns, the ip is not refreshed in local (computer, ipad etc)
5) tried youtube as example, where ttl is 5 mins, but still not refreshed after this

Possible solution:
Give option to add —max-cache-ttl in dnsmasq startup parameter to allow set ttl to 60 or 0 (small number)
Right now this option is not available in dnsmasq.conf and no access to change startup parameter on dnsmasq

Thanks.

RMerl, 2019-07-24:
Dnsmasq cache is not persistent.  So if you still have entries cached after restarted dnsmasq, then your cache lies somewhere else - quite probably at your client OS level.

Test with dig to investigate the actual TTL value of returned results.

doracat55555, 2019-07-24:
Thanks for the suggestion. I tested again on the router ssh itself, the issue appears to be the dns provider (cleanbrowsing in this case) returns different response when doing normal DNS (port 53) or DNS over TLS (port 853). The normal DNS returns correct response, but DNS over TLS returns unfiltered result. I changed to normal DNS on router, and issue resolved immediately.

I have hard time to clean up DNS cache on the router itself, maybe convenient to have some script that clean up the cache on the router.

RMerl, 2019-07-24:
There is no cache to clear on the router.  Dnsmasq keeps it in RAM, so simply restarting dnsmasq will clear it.  Stubby itself does not cache results.

doracat55555, 2019-07-24:
Got it. Thank you for the response.

closed 2019-07-24 by doracat55555