RMerl/asuswrt-merlin.ng #338: asuscomm.com DDNS Can't Get Right External IP in China

yykhysn, 2019-07-24:
For "checkip.dyndns.org" is not available in China, PLEASE update the API to others like "checkip.dyndns.com". 

Maybe it will cost a long time to release the new firmware. Could tell me how can I change the API personally or how can I redirect the url "checkip.dyndns.org" to "checkip.dyndns.com" on router.

Desperately in need of help,
Very Thankful!

RMerl, 2019-07-24:
Set mode to "Internal" instead of External, it will use the IP from nvram instead of an external query.

RMerl, 2019-07-24:
Note that checkip.dyndns.org is the same thing as checkip.dyndns.com - it's just a CNAME to it.  See the first line in the ANSWER section:

```
merlin@ubuntu-dev:~$ dig checkip.dyndns.org

; <<>> DiG 9.11.3-1ubuntu1.8-Ubuntu <<>> checkip.dyndns.org
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 45344
;; flags: qr rd ra; QUERY: 1, ANSWER: 5, AUTHORITY: 0, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 65494
;; QUESTION SECTION:
;checkip.dyndns.org.		IN	A

;; ANSWER SECTION:
checkip.dyndns.org.	479	IN	CNAME	checkip.dyndns.com.
checkip.dyndns.com.	479	IN	A	216.146.43.70
checkip.dyndns.com.	479	IN	A	216.146.43.71
checkip.dyndns.com.	479	IN	A	131.186.113.70
checkip.dyndns.com.	479	IN	A	162.88.193.70

;; Query time: 0 msec
;; SERVER: 127.0.0.53#53(127.0.0.53)
;; WHEN: Wed Jul 24 02:27:45 EDT 2019
;; MSG SIZE  rcvd: 143
```

commit 208dea49 (2019-07-24): Change default DynDNS IP server to the .com TLD, since .org cannot be resolved in China (closes #338)

Since the .org hostname is just a CNAME to the .com one, this should have no functionality change,
except allow Chinese users to also use it.
closed 2019-07-24 by RMerl