RMerl/asuswrt-merlin.ng #344: Some images can't be found

geoleonsh, 2019-08-11:
Example:
asuswrt-merlin.ng/release/src/router/www/images/New_ui/hightaba.png
asuswrt-merlin.ng/release/src/router/www/images/New_ui/inputbg.png
asuswrt-merlin.ng/release/src/router/www/images/New_ui/inputbg_2.png
asuswrt-merlin.ng/release/src/router/www/images/New_ui/inputbg_disable.png

In the lastest source,these images don't exist

RMerl, 2019-08-14:
I don't understand your point.  These files don't; exist, and there's no reference to them either in the web pages, so I don't understand why you expect these particular images to exist.

geoleonsh, 2019-08-15:
I got some error message in the web pages，about images 404 error.
But it returned to normal after a while.
I opened the issues before it returned to normal.

RMerl, 2019-08-15:
You probably had old content in your browser cache which needed to expire.

closed 2019-08-15 by RMerl