RMerl/asuswrt-merlin.ng #345: Can't upgrade to 384.13

surdu, 2019-08-13:
I have an RT-AC68U, and I tried to manually update from 384.12 to 384.13, but after I reboot, the version is still 384.12 and the update notification still there.

marodor, 2019-08-13:
Try this:
Remove usb devices hooked on the router, reboot, and retry flashing

surdu, 2019-08-13:
@marodor Thank you for the tip, but unfortunately didn't worked

asaddon, 2019-08-15:
Is it a genuine AC68U or a converted T-mobile model?

surdu, 2019-08-15:
Genuine. Wet don't even have T-mobile where I'm from.

asaddon, 2019-08-15:
Airtight then try power cycling the router and then use a different browser to do the upgrade this time.

RMerl, 2019-08-20:
The issue tracker is only for development purposes.  If you need technical support or have a question, please use the support forums at https://www.snbforums.com/forums/asuswrt-official.51/ instead.

closed 2019-08-20 by RMerl

surdu, 2019-08-25:
@asaddon That didn't work either. Is there a log somewhere that keeps tabs of what's going on during upgrade?

@RMerl This sounds to me like there is an issue with the upgrade process in asuswrt-merlin. This is an issue tracker, right?

asaddon, 2019-08-26:
I'm not aware of any update logs in the  syslog. The only option left now is do a complete factory reset and then try to update again.