RMerl/asuswrt-merlin.ng #347: there is pink page with error in aiprotection - yandex dns

YBS-PC, 2019-08-15:
[image]
[image]

_Originally posted by @YBS-PC in https://github.com/RMerl/asuswrt-merlin.ng/issues/346#issuecomment-521601261_

YBS-PC, 2019-08-15:
after all three cases of firmware update: update, nvram clean & update, restoration

RMerl, 2019-08-15:
That tab doesn't show here.  If it shows on your router, then something else is interfering with the webui from properly showing/hiding tabs.

Try a different browser, also look at your browser console for any Javascript error preventing the tab from being properly removed.  Also if you run any add-on that modifies the webui, disable those add-ons.

YBS-PC, 2019-08-15:
this tab is in the firmware from ASUS (perhaps for Russia, but I flashed the firmware from the international site and from the Hong Kong site and there it is also), but in the Merlin firmware the tab is also there, but it is not functional and leads to an error page. Checked on a new clean browser Mozila, chrome and Opera

RMerl, 2019-08-15:
That would imply that this tab is shown when the router`s region is set to RU in the bootloader.  @themiron , can you confirm if this is indeed the case?  If so, they I will need to take active steps to hide it, as it might interfere with the DNSFilter implementation.

If you need the YandexDNS functionality then you can reproduce it by using DNSFilter, on the LAN section.

RMerl, 2019-08-20:
I can't see any evidence in the code that would show that this tab wouldn't be removed unless the firmware was compiled with explicit YandexDNS support.  Please post a screenshot of your Tools -> Sysinfo page, showing the current Features flags.

YBS-PC, 2019-08-20:
[image]

RMerl, 2019-08-20:
Your feature list reports that yadns is enabled, yet I did not enable it when I built the RT-AC88U firmware.  I also see you have an unexpected bootloader version, so I suspect that maybe the Russian SKU of the RT-AC88U has an hardcoded list of features that enables YandexDNS regardless of whether the firmware supports it or not.

Only way I can see to fix this would be to forcibly remove the YandexDNS tab from the webui regardless of the state of the yadns feature flag.

YBS-PC, 2019-08-20:
> Only way I can see to fix this would be to forcibly remove the YandexDNS tab from the webui regardless of the state of the yadns feature flag.

How can i do this or it will be in new release?

> unexpected bootloader version

Can i change it?

RMerl, 2019-08-20:
Just don't click on the tab...  The fix will simply hide the tab from the webui, there will be no change in functionality.

The bootloader cannot be changed by the end-user.

commit 7f88f1f8 (2019-08-28): webui: ensure that YandexDNS is always disabled at the webui level (closes #347)

Russian SKUs would enable it despite firmware being compiled without
YandexDNS support, leading to a link to a non-existant (and non-
functional page) to appear on the webui.  DNSFilter will cover the
same functionality but with a broader scope.
closed 2019-08-28 by RMerl