RMerl/asuswrt-merlin.ng #348: Adaptive QoS Tracking: tooltips incorrect

grnassar, 2019-08-18:
With Adaptive QoS enabled, Adaptive Qos->Classification tab->Tracked Connections: the Application column, the tooltips that appear on hovering over an application name are mostly incorrect. "Net Control Packets" and "Default" show up correctly, but all others show something other than their category (usually the category below theirs; for example, the tooltip for an application in the VoIP category shows up as "Gaming.")

Screenshot of bug:
https://postimg.cc/34SC53MD

RMerl, 2019-08-20:
All tooltips are showing correctly for me.  I don't see anything wrong on your screenshot either, you have an application connected to an Amazon server that reports that it's doing a file transfer, which doesn't seen unusual to me.

grnassar, 2019-09-06:
Strangely, I started using FreshJR's script after that and it *fixed* the tooltips for me.  I can remove it again and see if the problem returns.  Perhaps just some fluke one-off, I guess.

closed 2020-09-21 by RMerl