RMerl/asuswrt-merlin.ng #350: feature: hide/disable aimesh (to suppress its side-effects, like additional modal alerts)

bughit, 2019-08-20:
Would be nice to able to disable/hide aimesh, in the manner it was hidden/disabled by default in prior versions. I am assuming this would disable aimesh specific alerts, like radio off confirm.

Those who don't need aimesh, which I would guess is a significant %, probably don't want any changes it brings, especially additional modal alerts, and would be happy to hide/disable it.

bughit, 2019-09-21:
@RMerl Does stock make it easy to hide aimesh? Seems like it should since it has to support models without it. If so would you consider this request?

RMerl, 2019-09-21:
It's a compile time option, something only Asus can enable/disable since it includes closed source portions of the code.

bughit, 2019-09-22:
> It's a compile time option

You are saying you could only (in theory) produce a build with it off, but can't make it toggleable by the user at run-time?

> something only Asus can enable/disable

I don't understand "only asus". Your builds had it hidden for a time, so you must have been using this compile-time option?

RMerl, 2019-09-22:
Fully enabling and disabling can only be done by Asus.

Hiding it at the webui level was a hardcoded hack I had in place temporarily, and would make no sense as a user toggleable feature, since all it did was hide the settings from the webui.  If you don't want to use Aimesh, simply don't use it, same as with any other router feature.

bughit, 2019-09-22:
>  If you don't want to use Aimesh, simply don't use it

I am forced to "use it" every time I toggle the radio (#342).

> would make no sense as a user toggleable feature, since all it did was hide the settings from the webui

I am fairly sure the hiding hack disabled aimesh related side-effects like #342, which is why I requested it as an alternative to the rejected #342.

closed 2020-09-21 by RMerl