RMerl/asuswrt-merlin.ng #354: Adaptive QoS mode configuration lost after reboot

reejue, 2019-09-02:
**This is a bug since 384.12_0 to 384.13**
 384.11_2 and the official firmware works well.

I set the Adaptive QoS mode to Games, applied successful.
Then **reboot** router , it has been lost and shows Customize mode icon insteand.

Thank you RMerl help me fix this.

Screenshot:
http://chuantu.xyz/t6/702/1567417474x992245975.png

reejue, 2019-09-02:
And the Game Boost - One Click Game First mode button trun to **OFF** status.

Screenshot: 
[image]

RMerl, 2019-09-02:
The bug lies in the WLCSM component, which is closed source.  Asus are aware of the issue and should have fixed it only recently.  Until they release a new firmware with the fix there is nothing I can do.

closed 2019-09-02 by RMerl