RMerl/asuswrt-merlin.ng #356: Asus Rapture AX11000 support?

breisig, 2019-09-05:
Are their any plans to support the Asus RoG Rapture AX11000 802.11ax access point?

RMerl, 2019-09-05:
No plan at this time. I don`t support ROG models due to the increased workload in maintaining their customized web interface in addition to the regular one.

closed 2019-09-05 by RMerl

REZISTENCE, 2020-06-20:
Hi there
Just on this topic near one year on, would you be looking into this or think you might want to give the ROG a go? Understanding there is a higher workload involved, but there is most likely a greater user base now too. 

Why do ROG users keep asking? Cause we have seen your work on the other models and your stuff is just better than ASUS.com. 

Thank you for all your work on the other ASUS models.

onek24, 2020-06-20:
^+1
I would be interested as well

lernamanto, 2020-06-27:
+1
I would be very interested as well!!

Thank you for all your work on all other models

Jayonics, 2020-07-11:
+1 
I have spent the last 2 years using Windows Server for Active Directory and trying to get it to play nicely alongside this router but it always so much hassle with the ASUS stock firmware and the way they have constructed the OS. Whenever I have something I try to do and search '[Insert task here] Asus router' I always end up coming across a page by either Openwrt or Merlin and is something done on a single page.

But I spend weeks on a single task because of how restricted the OS is when you finally SSH in and it overwrites configurations, ignores scripts and has such a limited range of packages despite being a quad-core mini computer. Please help us peps out :(

waldemar-szostak, 2020-11-05:
I would also very much like Merlin available for this router. I'm sure there are others like myself who will gladly to contribute financially then.

endtimeszero, 2020-11-19:
I would love to see this supported as well. Merlin's work is amazing and he's always been responsive to all of my questions.

hurda, 2020-11-20:
If the Merlin-fw is so important for you, why are you buying hardware which isn't supported?

Jayonics, 2020-11-20:
> If the Merlin-fw is so important for you, why are you buying hardware which isn't supported?

What a pointless question...

It's like saying, 'if 32 bit programs only worked on 32 bit OS'es, why would you buy a 64 bit OS'

It doesn't take a genius to figure out the 2 possible answers to your question:
1. You didn't know about custom firmware until after buying the unsupported devices.
2. The devices have better hardware.

The whole point of this thread is people pointing out the newer devices are growing in popularity ( The main reason they claimed to not support it ) and that people want this highly desirable custom firmware.

Don't post more non-scensicle questions, please.

hurda, 2020-11-20:
RMerlin's reply in this ticket doesn't even mention popularity. 
Neither is it mentioned in his FAQs in the snbforums: 
https://www.snbforums.com/threads/faq-read-me-first-before-posting-a-question.47153/#post-609193

To make things short:
`So don't ask if/when a new model will be supported. I cannot speculate about future model support, and until I have working support for a model, I cannot answer that question.`

REZISTENCE, 2020-11-27:
Are you simple?

Plenty of people would like to see Merlin offered on this product. I am far from the only one that has expressed interest in Merlin for this model. Who are you to me this? If people don’t express interest then nothing would eventuate, so thank you for trying to suppress my wants and needs but this is my life and I don’t feel that it hurts to my interest forward as in doing so, so have others and perhaps that might help. Anyways, if you are not interested in the product getting Merlin butt out, as it is of no interest to you.

Stop being so arrogant

REZISTENCE, 2020-11-27:
Actually Hurda it does mention popularity?

Quote
Q: Will you support the GT-xxxxx?
A: No, because of infrequent GPL releases from Asus (mandatory for proper support by Asuswrt-Merlin), the amount of extra work involved in supporting specifically those GT models, and their low popularity due to their high price.

And as for price people are realising that they have wants and needs for this hardware in the home in first world countries now and can afford such devices if one device will cover all there needs. It is becoming a question of coverage in a household.

I think you will be surprised how many people would actually be interested but just have not bought it due merlin not supporting it.

If it is a case of money for the merlin developer, pop up a fund for it and I am sure you will get people donating to this cause. Why not what has merlin got to lose y supporting the top model.

From: נυѕт נ<mailto:reply2menot@outlook.com>
Sent: Friday, 27 November 2020 11:59 PM
To: RMerl/asuswrt-merlin.ng<mailto:reply@reply.github.com>
Subject: RE: [RMerl/asuswrt-merlin.ng] Asus Rapture AX11000 support? (#356)

Are you simple?

Plenty of people would like to see Merlin offered on this product. I am far from the only one that has expressed interest in Merlin for this model. Who are you to me this? If people don’t express interest then nothing would eventuate, so thank you for trying to suppress my wants and needs but this is my life and I don’t feel that it hurts to my interest forward as in doing so, so have others and perhaps that might help. Anyways, if you are not interested in the product getting Merlin butt out, as it is of no interest to you.

Stop being so arrogant