RMerl/asuswrt-merlin.ng #363: DNSFilter "No Filtering" not working

YBS-PC, 2019-09-19:
on the tab "DNS filter" it is written that "No Filtering" will disable/bypass the filter, and "Router" will force clients to use the DNS provided by the router's DHCP server (or, the router itself if it's not defined)".
But if I specify in "DNS Server 1" any server, and I specify for the client "without filtering" in "DNS filter", it all the same passes DNS requests through the server specified in "DNS Server 1"

YBS-PC, 2019-09-19:
UPD
if I specify in "DNS Server 1" any server, and I specify for the client any other server in "DNS filter", any way it all the same passes DNS requests through the server specified in "DNS Server 1"

dave14305, 2019-09-19:
You can get user support at https://www.snbforums.com/forums/asuswrt-merlin.42/

TheGITofTeo997, 2019-12-08:
I have also been experiencing this with my Raspberry, the No Filtering option is not being respected

closed 2020-09-21 by RMerl