RMerl/asuswrt-merlin.ng #364: Update nano to v. 4.5

feranick, 2019-10-08:
Changelog
The new 'tabgives' command allows you to specify per syntax whatthe <Tab> key should produce: some spaces, a hard TAB, ...The output of --help is properly aligned again for all languages.<Tab> will indent a marked region also when M-} has been rebound.
--

feranick, 2019-12-02:
Version 4.6 is available
Changelog
The 'formatter' command has returned, bound by default to M-F.It allows running a syntax-specific command on the contents ofthe buffer.^T will try to run 'hunspell' before 'spell', because it checksspellling for the locale's language and understands UTF-8.Multiple errors or warnings on startup will no longer slow nanodown but will be indicated on the status bar with trailing dots.
--

feranick, 2019-12-24:
Version 4.7 is available
Changelog
A <Tab> will indent a marked region only when mark and cursor are on different lines.
Two indentations (any mix of tabs and spaces) are considered the same when they look the same (that is: indent to the same level).
When using --breaklonglines or ^J, a line will never be broken in its leading whitespace or quoting.
The keywords in nanorc files must be in lowercase.

feranick, 2020-02-01:
Any specific reason this is not updated in the latest beta for 384.15?

RMerl, 2020-02-02:
It's just not a priority, I only occasionally update nano.  There is little incentive to keep it up-to-date, since it only gets used for occasional script editing, therefore the vast majority of the features aren't even used.

commit c4bc78ed (2020-02-10): nano: updated to 4.8 (closes #364)
closed 2020-02-10 by RMerl