RMerl/asuswrt-merlin.ng #367: Option "server" in /jffs/configs/dnsmasq.conf.add have no effect

Myppomeu, 2019-10-13:
### Description

I have my home DNS zone, "home". It served by server in my LAN.
After changing router settings according to [this manual](https://github.com/RMerl/asuswrt-merlin/wiki/Custom-domains-with-dnsmasq) got something like this:

```
user@router:/tmp/home/root# cat /jffs/configs/dnsmasq.conf.add
server=/home/192.168.1.10
```

Router was restarted and file ```/jffs/configs/dnsmasq.conf.add``` persisted.

After configuration all ```home``` zone is not resolvable from routers client PCs. It is not resolvable even from router. I expected name resolution in ```home``` zone will work transparently.

That's what I got from client PC with dig (```192.168.1.1``` is router):

```
dig @192.168.1.1 server.home.

; <<>> DiG 9.11.5-P4-5.1-Debian <<>> @192.168.1.1 server.home.
; (1 server found)
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 32942
;; flags: qr aa rd ra; QUERY: 1, ANSWER: 0, AUTHORITY: 0, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 4096
;; QUESTION SECTION:
;server.home.               IN      A

;; Query time: 127 msec
;; SERVER: 192.168.1.1#53(192.168.1.1)
;; WHEN: Вс окт 13 04:44:23 +05 2019
;; MSG SIZE  rcvd: 44
```

DNS server at that moment worked good and ansqered to requests.

### Info

Router model: **ASUS RT-AC68U**
Asuswrt-merlin.ng version: **384.13_0**

### Additional info

Such config worked as expected in 384.7
After upgrading from 384.7 to 384.13_0 I erased nvram with ```mtd-erase2 nvram``` (web-interface was closed) and phisically restarted router.

Is there other information I can provide?

asaddon, 2019-10-13:
I believe you're using a T-Mobile converted AC68U, if yes then it's not officially supported on AsusWRT-Merlin.

Myppomeu, 2019-10-13:
> using a T-Mobile converted AC68U

How can I found it? I have a router box, there is no mention about T-Mobile. A1 hardware manufactured in 2014.

> it's not officially supported

I think it's software problem since in 384.7 "server" option worked perfect on the same router.

asaddon, 2019-10-13:
It's just a guess but if you can do the firmware upgrades directly from WebUI then you're on retail AC68U.

closed 2020-09-21 by RMerl