RMerl/asuswrt-merlin.ng #371: AC68U Network map clients not loading.

MertYuce, 2019-10-21:
Cannot see a list of online clients in Network Map. I also included some other warnings from the Chrome Console which happens only when i visit Adaptive QoS page. I'm not sure if they are related because i don't see those warnings at the Network Map page.

This problem has been fixed in the non-wrt firmware (tried, it works);
Version on Asus.com
> Version 3.0.0.4.384.81049
> 2019/09/05
> Bugfix
> - Fixed Network map client list issues.

Console log on chrome
```
jquery.js:5 [Deprecation] Synchronous XMLHttpRequest on the main thread is deprecated because of its detrimental effects to the end user's experience. For more help, check https://xhr.spec.whatwg.org/.
send @ jquery.js:5
AdaptiveQoS_Bandwidth_Monitor.asp:1 A cookie associated with a cross-site resource at https://www.asus.com/ was set without the `SameSite` attribute. A future release of Chrome will only deliver cookies with cross-site requests if they are set with `SameSite=None` and `Secure`. You can review cookies in developer tools under Application>Storage>Cookies and see more details at https://www.chromestatus.com/feature/5088147346030592 and https://www.chromestatus.com/feature/5633521622188032.
AdaptiveQoS_Bandwidth_Monitor.asp:1 A cookie associated with a cross-site resource at http://asus.com/ was set without the `SameSite` attribute. A future release of Chrome will only deliver cookies with cross-site requests if they are set with `SameSite=None` and `Secure`. You can review cookies in developer tools under Application>Storage>Cookies and see more details at https://www.chromestatus.com/feature/5088147346030592 and https://www.chromestatus.com/feature/5633521622188032.
AdaptiveQoS_Bandwidth_Monitor.asp:1 A cookie associated with a cross-site resource at http://www.asus.com/ was set without the `SameSite` attribute. A future release of Chrome will only deliver cookies with cross-site requests if they are set with `SameSite=None` and `Secure`. You can review cookies in developer tools under Application>Storage>Cookies and see more details at https://www.chromestatus.com/feature/5088147346030592 and https://www.chromestatus.com/feature/5633521622188032.
```

Mert-Y, 2019-12-15:
Fixed with version 384.14_0

closed 2019-12-15 by Mert-Y

jwarner112, 2019-12-18:
Dealing with this issue myself. Trying to update to `384.14_0` but unable to, for some reason. Will try again soon.