RMerl/asuswrt-merlin.ng #373: Remove or update acme-client-portable from code

doomfrawen, 2019-10-23:
Hi,
The acme-client-portable is no longer usable nor maintained. The API URI  has changed as well as their methods, therefore the client is unable to renew/request new certificates.

Old API: https://acme-v01.api.letsencrypt.org/
New API: https://acme-v02.api.letsencrypt.org/

The current client is giving the following errors: 

custom_script: Running /jffs/scripts/service-event (args: restart letsencrypt)
Oct 23 09:48:04 kernel: /usr/sbin/acme-client: SSL_read return 5: Success
Oct 23 09:48:04 kernel: /usr/sbin/acme-client: https://acme-v01.api.letsencrypt.org/acme/new-reg: bad comm
Oct 23 09:48:04 kernel: /usr/sbin/acme-client: transfer buffer: [{ "FRcprCYGxBA": "https://community.letsencrypt.org/t/adding-random-entries-to-the-directory/33417", "key-change": "https://acme-v01.api.letsencrypt.org/acme/key-change", "meta": { "caaIdentities": [ "letsencrypt.org" ], "terms-of-service": "https://letsencrypt.org/documents/LE-SA-v1.2-November-15-2017.pdf", "website": "https://letsencrypt.org" }, "new-authz": "https://acme-v01.api.letsencrypt.org/acme/new-authz", "new-cert": "https://acme
Oct 23 09:48:05 kernel: /usr/sbin/acme-client: SSL_read return 5: Success
Oct 23 09:48:05 kernel: /usr/sbin/acme-client: https://acme-v01.api.letsencrypt.org/acme/new-authz: bad comm
Oct 23 09:48:05 kernel: /usr/sbin/acme-client: transfer buffer: [{ "FRcprCYGxBA": "https://community.letsencrypt.org/t/adding-random-entries-to-the-directory/33417", "key-change": "https://acme-v01.api.letsencrypt.org/acme/key-change", "meta": { "caaIdentities": [ "letsencrypt.org" ], "terms-of-service": "https://letsencrypt.org/documents/LE-SA-v1.2-November-15-2017.pdf", "website": "https://letsencrypt.org" }, "new-authz": "https://acme-v01.api.letsencrypt.org/acme/new-authz", "new-cert": "https://acme

More info  about the topic: https://community.letsencrypt.org/t/certificate-not-updating-anymore-asus-router/103647

From a security point of view, I would recommend removing this client and notifying the user on the GUI until a new client could be ported.

RMerl, 2019-10-23:
Asus are already working on updating the client on their end, so any update will be implemented once they release a new firmware using it.

doomfrawen, 2019-10-23:
Thanks for the quick response. I thought it was more of a Merlin issue. I was getting ready to port some clients :)

RMerl, 2019-10-23:
The client they use is no longer actively developed (the author only maintains the native BSD client nowadays), so I don't know if they will replace the client with something else (I wonder if acme.sh would work with the Busybox shell), or backporting the V2 support from the BSD client.

In any case, the library calling the client is closed source, so any client change will have to be done by Asus.

jorhett, 2019-11-26:
FWIW I've found the dehydrated client to work well, and all of it's requirements are available on the node https://github.com/lukas2511/dehydrated

If we have to wait too long for Asus, creating a wrapper script that calls dehydrated should be pretty easy.

RMerl, 2019-11-26:
Asus already fixed it by implementing acme.sh.

closed 2019-11-26 by RMerl

maltokyo, 2019-12-01:
Hi @RMerl is there a new version of FW with this fixed?  I cant login due to SSL cert errors.

praveen12bnitt, 2019-12-03:
Any plans to release a newer version of Merlin with this fix in ? I am sure a lot of people like me with expired certs are not able to login from outside wifi

RMerl, 2019-12-03:
The fix is already present in 384.14 Beta 2.

maltokyo, 2019-12-03:
Indeed. I got the beta yesterday and it fixes this.

praveen12bnitt, 2019-12-03:
Thanks guys. i will take the beta release.

praveen12bnitt, 2019-12-05:
Installed the beta last night and everything started working again.. Thanks!

tdensmore, 2019-12-28:
Im still having trouble. I have firmware 384.14_beta3 running on my RT-AC3100 but Im still stuck on "updating" ...

RMerl, 2019-12-28:
Use the forums if you need tech support.  LE is definitely fixed with 384.14, so your issue is elsewhere.