RMerl/asuswrt-merlin.ng #376: IPv6 DNS Servers not applied to dnsmasq configuration

aaronwmorris, 2019-10-30:
The IPv6 DNS servers under the IPv6 DNS Setting are not being applied to the dnsmasq configuration.

No matter what is entered for the IPv6 DNS settings, the following is always added to the dnsmasq.conf

`dhcp-option=lan,option6:23,[::]`

Device:  RT-AC68U
Version:  384.13

Dabombber, 2019-11-18:
DNS servers are added to the file _/tmp/resolv.dnsmasq_, and read by dnsmasq with the line
`servers-file=/tmp/resolv.dnsmasq`
The dhcp-option is to tell LAN clients to use the router as their DNS server.

kevinpark1217, 2020-03-22:
Is there any workaround or plans to fix this issue?

closed 2020-09-21 by RMerl

drone540, 2021-06-14:
I posted a solution to this here: https://www.snbforums.com/threads/send-ipv6-lookups-to-pihole.72595/#post-694573

The problem seems to be that the DNS Server fields on the IPv6 page does not set the:
dhcp-option=lan,option6:23,[The:DNS:That:You:Set:Here]

The setting of [::] ends up contacting your ISP based on the WAN setting unless you set it to Connect to DNS Server Automatically No.