RMerl/asuswrt-merlin.ng #381: Question: Did you write this code base completely from scratch, or did you use asuswrt as a starting point?

ScottPierce, 2019-11-24:
Did you write this code base completely from scratch, or is it somehow tweaked from the Asuswrt software, using it as a base? I'm curious because I'm trying to evaluate the amount of risk associated with me using this project. I've read the `about` section etc it mentions that you add a few few features etc, but it never clearly states if it's completely written from scratch, etc, and I haven't been able to find an open source version of asuswrt.

RMerl, 2019-11-24:
It's based on Asuswrt.  New GPL releases get merged on top of it as they are released by Asus.  Those GPL drops can be downloaded from their support site for the specific model you are looking at.

ScottPierce, 2019-11-25:
Awesome - that significantly lowers the risk of using this.

closed 2019-11-25 by ScottPierce