RMerl/asuswrt-merlin.ng #383: Send MagicPacket with shell

Catskan, 2019-11-26:
Hi guys,

I have a Asus RT-AC86U with AsusWRT-Merlin on it and I want to send and MagicPacket with the ssh.
When I use the WebUi to send this packet, it works but I doesn't work with the ssh.

I'd tryied :
/usr/sbin/ether-wake -i br0 MM:AA:CC:AA:DD:RR
and
/usr/sbin/ether-wake -i br0 -b MM:AA:CC:AA:DD:RR

In the system log, I can't see what is command send.

Thanks you in advance.

Regards

crkinard, 2019-11-26:
/usr/sbin/ether-wake -i br0 MM:AA:CC:AA:DD:RR

Yeah that is exactly what I use in a script loaded into cron to wake my desktop just before I get home every day. On a 86 as well.

Catskan, 2019-11-27:
Hi,

I retried this command but doesn't work.

In my ssh terminal, I start this command but my PC doesn't wake up.
But, with the GUI, my PC wake up immediatly

Catskan, 2019-12-19:
Hi,

So, I retried this command and works now.

Thanks you for it.

Catskan

closed 2019-12-19 by Catskan