RMerl/asuswrt-merlin.ng #385: RFC 4638 - ASUS RT-AX88U PPPoE 1500 doesnt work

jonathanmasseurca, 2019-11-28:
I'm using Bell Fiber (PPPoE on vlan 35)

I tried to set MTU and the MRU to 1500 (Wan PPPoE settings) and inside my log file i have those errors 

May  5 01:05:10 start_wan_if(): Error setting MTU on eth0 to 1500
May  5 01:05:10 start_wan_if(): Error setting MTU on vlan35 to 1500

Bell is RFC 4638 and on my old router AC3100 i dont have this problem

I cannot do a ping google.ca -f -l 1472 without being fragmented on the ASUS RT-AX88U, only 1464 works which means that the MTU is set to 1492.

jonathanmasseurca, 2019-11-28:
br0       Link encap:Ethernet  HWaddr 0C:9D:92:03:A7:F0
          inet addr:192.168.1.1  Bcast:192.168.1.255  Mask:255.255.255.0
          UP BROADCAST RUNNING ALLMULTI MULTICAST  MTU:1500  Metric:1
          RX packets:1116882 errors:0 dropped:14 overruns:0 frame:0
          TX packets:6502642 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:115861801 (110.4 MiB)  TX bytes:9715325463 (9.0 GiB)

eth0      Link encap:Ethernet  HWaddr 0C:9D:92:03:A7:F0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:6470502 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1095077 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:9760382571 (9.0 GiB)  TX bytes:141508053 (134.9 MiB)

eth1      Link encap:Ethernet  HWaddr 0C:9D:92:03:A7:F0
          UP BROADCAST ALLMULTI MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth2      Link encap:Ethernet  HWaddr 0C:9D:92:03:A7:F0
          UP BROADCAST ALLMULTI MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth3      Link encap:Ethernet  HWaddr 0C:9D:92:03:A7:F0
          UP BROADCAST RUNNING ALLMULTI MULTICAST  MTU:1500  Metric:1
          RX packets:1105783 errors:0 dropped:11 overruns:0 frame:0
          TX packets:6503960 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:128005061 (122.0 MiB)  TX bytes:9716600139 (9.0 GiB)

eth4      Link encap:Ethernet  HWaddr 0C:9D:92:03:A7:F0
          UP BROADCAST RUNNING ALLMULTI MULTICAST  MTU:1500  Metric:1
          RX packets:31546 errors:0 dropped:51 overruns:0 frame:0
          TX packets:25041 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:4892555 (4.6 MiB)  TX bytes:2572862 (2.4 MiB)

eth5      Link encap:Ethernet  HWaddr 0C:9D:92:03:A7:F0
          UP BROADCAST RUNNING ALLMULTI MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:10686 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:2262814 (2.1 MiB)

eth6      Link encap:Ethernet  HWaddr 0C:9D:92:03:A7:F0
          UP BROADCAST RUNNING ALLMULTI MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:15 overruns:0 frame:0
          TX packets:6113 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:1273178 (1.2 MiB)

eth7      Link encap:Ethernet  HWaddr 0C:9D:92:03:A7:F4
          UP BROADCAST RUNNING ALLMULTI MULTICAST  MTU:1500  Metric:1
          RX packets:21376 errors:0 dropped:15 overruns:0 frame:0
          TX packets:27100 errors:0 dropped:6 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:1583729 (1.5 MiB)  TX bytes:4154009 (3.9 MiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          UP LOOPBACK RUNNING MULTICAST  MTU:65536  Metric:1
          RX packets:6576 errors:0 dropped:0 overruns:0 frame:0
          TX packets:6576 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:1494902 (1.4 MiB)  TX bytes:1494902 (1.4 MiB)

lo:0      Link encap:Local Loopback
          inet addr:127.0.1.1  Mask:255.0.0.0
          UP LOOPBACK RUNNING MULTICAST  MTU:65536  Metric:1

ppp0      Link encap:Point-to-Point Protocol
          inet addr:[HIDDEN]  P-t-P:10.11.18.121  Mask:255.255.255.255
          UP POINTOPOINT RUNNING MULTICAST  MTU:1492  Metric:1
          RX packets:6470456 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1095021 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:3
          RX bytes:9592132058 (8.9 GiB)  TX bytes:106965399 (102.0 MiB)

spu_ds_dummy Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00
          UP RUNNING NOARP  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:100
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

spu_us_dummy Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00
          UP RUNNING NOARP  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:100
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

vlan35    Link encap:Ethernet  HWaddr 0C:9D:92:03:A7:F0
          inet addr:169.254.236.41  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST RUNNING ALLMULTI MULTICAST  MTU:1500  Metric:1
          RX packets:6470502 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1095077 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:9643897739 (8.9 GiB)  TX bytes:131055633 (124.9 MiB)

jonathanmasseurca, 2019-11-29:
I made another test, the jumbo frame also doesnt work, even if it's enabled

admin@RT-AX88U:/tmp/home/root# ifconfig eth0 mtu 1508 up
ifconfig: SIOCSIFMTU: Invalid argument

RMerl, 2019-11-29:
Anything tied to the switch requires the proprietary documentation from Broadcom to know how to configure, so it's out of my control.

closed 2020-09-21 by RMerl