RMerl/asuswrt-merlin.ng #388: AX88U: 384.14 beta 2 -  Cannot change System settings.

nievesj, 2019-12-01:
AX88U
Firmware Version:384.14_beta2-new-gfec6767101

Apply button on Administration -> System doesn't work. Firefox throws the following error:

```
TypeError: document.form.sshd_port_x is undefinedAdvanced_System_Content.asp:646:4
    validForm http://192.168.50.1/Advanced_System_Content.asp:646
    applyRule http://192.168.50.1/Advanced_System_Content.asp:304
    onclick http://192.168.50.1/Advanced_System_Content.asp:1
```
The variable ssh_port_t is always undefined.

Tried clean install, clearing nvram, factory default initialize, and the error persists.

RMerl, 2019-12-01:
I know, already fixed internally.

Don't report bugs about this special test build, it's not intended for release, just for specific testing.

closed 2019-12-01 by RMerl

nievesj, 2019-12-01:
Is there a guideline to determine where bugs from special builds should be reported? Looked through the tickets and there was no other ticket or way to know it was a known issue.