RMerl/asuswrt-merlin.ng #389: Iimprovement] Asus RT-AX88U 3.0.0.4.384.7756

lorenc-tomasz, 2019-12-01:
Add updates from latest firmware if possible: 
- Enhanced the Wi-Fi 6 performance with Apple® iPhone11, Samsung® S10.
- Supported WPA3
- Supported OFDMA
- Supported 802.11k and 802.11v
- Fixed Let's encrypt register related bugs.
- Fixed connection issues between Android and RT-AX88U IPSec VPN server.
- Improved system stability.

RMerl, 2019-12-02:
The GPL was already merged nearly a week ago.

closed 2019-12-02 by RMerl

lorenc-tomasz, 2019-12-02:
@RMerl I don't see WPA3 option on AX router.

RMerl, 2019-12-02:
The merge was done after the Beta 2 release, the code is in the 7756 code branch.

That code branch hasn't been merged into the production rtax88 branch because it's too buggy (Adaptive QoS is completely broken in 3.0.0.4.384_7756), therefore 384.14 will be based on 3.0.0.3.384_6436, which does not use the newer (broken) SDK.

lorenc-tomasz, 2019-12-03:
@RMerl Thank you for the answer :) 👍