RMerl/asuswrt-merlin.ng #391: OpenVPN Client Policy Routing malfunctioning

TheGITofTeo997, 2019-12-08:
Hello, first i am sorry if this is a duplicate issue or if it has already been discussed.

So i am experiencing a Policy Routing malfunctioning under VPN Client, policies are sometimes not being respected (maybe after reboot). 
If i just click Apply without changing anything things go okay, i don't know if i am missing something.

Thanks for the work and the support

conorriches, 2020-04-09:
I too am experiencing similar behaviour however I don't know if the causes are the same, and I'm using a different build.

I've created an issue on the build I'm using here https://github.com/gnuton/asuswrt-merlin.ng/issues/34 

Could you let me know if this looks similar to what you're experiencing? Do you have any logs from when this error occurs? Did you manage to fix it?

TheGITofTeo997, 2020-04-21:
Actually, so much time has passed and i do not even remember the circumstances under which this was happening. What i can tell you isthat now i am running the latest build which is 384.16 and for now i am not encountering any policy routing issue.

closed 2020-09-21 by RMerl