RMerl/asuswrt-merlin.ng #393: Asus RT-AX88U flashed with Asuswrt-Merlin firmware 384.14 is not advertising the 5GHz wireless network

mxplea10, 2019-12-17:
I flashed a new Asus RT-AX88U router from Asus Firmware Version 3.0.0.4.384_7756 to Asuswrt Firmware Version 384.14.  Since flashing to the Asuswrt Firmware, the 5GHz wireless network is no longer advertised.  The 2.4GHz is advertised and I can connect to it.  This is my first Asus router and I'm configuring it hoping to find a better VPN solution than my Linksys WRT3200ACM dd-wrt router.  Has anyone experience this problem and found a fix?  I've attached the wireless setting.

[image]
[image]

mxplea10, 2019-12-18:
After restoring the router a couple of times, the Asuswrt-Merlin Flashed Asus RT-AX88U is now advertising the 5GHz wireless network.  I took the following steps; 

Step 1 - Flashed back to Asuswrt FW Ver: 3.0.0.4.384_7756 from Asuswrt-Merlin 384.14. The 5GHz wireless network was being advertised.

Step 2 - Flashed Asuswrt-Merlin 384.13. The 5GHz wireless network wasn't being advertised.

Step 3 - Flashed back to Asuswrt FW Ver: 3.0.0.4.384_7756.

Step 3 - Then, flashed back to Asuswrt-Merlin 384.14. The 5GHz wireless network is now being advertised.

I didn't make any configuration changes. Can anyone explain why the 5GHz wireless network is working now?

asaddon, 2019-12-18:
Disable 802.11x/WiFi 6 mode and try again.

asaddon, 2019-12-18:
Ah okay it's working now, you can ignore my previous suggestion. Although I recommend you to still turn WiFi 6 mode off if there's no WiFi 6 devices in your network.

RMerl, 2019-12-18:
Please use the support forums at SNBForums for support requests.

The RT-AX88U wifi is working, it's what I use myself as my primary router.

closed 2019-12-18 by RMerl

mxplea10, 2019-12-18:
Although the 5GHz wireless network was advertised and accessible with the original Asus firmware, I disabled the 802.11x/WiFi 6 mode in the Asuswrt-Merlin firmware but the 5GHz wireless network still didn't work.  It didn't work until I performed the steps mentioned above.  Since I haven't found the cause for the 5GHz wireless network not being advertised, I'll probably see this problem again. Thanks for your suggestion.