RMerl/asuswrt-merlin.ng #397: Port Forwarding can'nt saved

panther7, 2019-12-23:
[image]

H3llK33p3r, 2020-01-23:
I think I already have a similiar issue when I was "not connected". In some case, I can see menus and options on the router because I have checked the "keep connection" on login. But changing settings do not work and most of the time the login page is displayed. When you try to forward the port were you already log or you had logged in just before the operation ?

RMerl, 2020-01-23:
This feature definitely works, there is something in your particular setup interfering with the webui.  I recommend using the official support forums at SNBForums.

closed 2020-01-23 by RMerl