RMerl/asuswrt-merlin.ng #400: Let's encrypt not updating?

LaMpiR, 2019-12-26:
I have noticed that my certificate has not updated automatically. 

Issued by :	Let's Encrypt Authority X3
Expires on :	2019/12/13

I have: RT-AC87U with 384.13_2. Any chance of triggering the update manually?

krnkris, 2019-12-27:
Just upgrade to 384.14. It is updated and works.
Please close this thread.

Neeham, 2020-01-03:
> I have noticed that my certificate has not updated automatically.
> 
> Issued by : Let's Encrypt Authority X3
> Expires on : 2019/12/13
> 
> I have: RT-AC87U with 384.13_2. Any chance of triggering the update manually?

I think one way to manually update is to click none then choose apply followed by Disable the DDNS Client and click apply once again. Afterward, simply revert back and see if it work.

LaMpiR, 2020-01-03:
I have tried that and I am stuck with this:

Status :	Updating...

jackyaz, 2020-01-05:
Update to 384.14

LaMpiR, 2020-01-06:
> Update to 384.14/384.14_2 depending on your model

The latest version for my model is already being used. 
RT-AC87U: 384.13_2

jackyaz, 2020-01-06:
Apologies! You'll have to wait for Asus to release an updated GPL for a fix - contact them if you haven't already.

RMerl, 2020-01-06:
384.13_2 contains the Let's Encrypt fix.  If you are still facing issues, I recommend posting on the support forums at SNBForums.

closed 2020-01-06 by RMerl

vladut123, 2020-01-06:
> Apologies! You'll have to wait for Asus to release an updated GPL for a fix - contact them if you haven't already.

How i can contact them?