RMerl/asuswrt-merlin.ng #404: Issue with AiMesh on 384.14/384.14_2 (AC88U) to add new node

lunarok, 2020-01-03:
Hi,

Last week I buy Lyra Mini to create an AiMesh network.
I was still on 384.13 and I could add my 2 nodes.
But as my ethernet plug was not good and I see that a firmware update was avalaible, I decide to delete the AiMesh created (remove the 2 nodes) and create it later after upgrade and fix my ethernet connexions.

I did it and then impossible to add the Lyra Mini. Connected through ethernet it always fails at the end with "impossible to connect to Aimesh node".
Try without ethernet same.
Both nodes same.
They are discovered but impossible to add.
I reset them at each try. I try also reboot the router, same.

Then I remember the 2 updates in the week I did 384.14 and _2.
So I dwld the 384_13 and try, and it works. I can add the Lyra nodes at the first try.

A little worried to upgrade now to 384.14_2. Do you need something else ?

LuisSerrano, 2020-01-30:
I have the same problem with Lyra, I downgrade the firmware to the version 384.13 and everything works ok.
With the latest version (384_14_2) the log showed errors and core 2 was always 100% working

B-Art, 2020-02-04:
Did you disable "AsusNAT"?

LuisSerrano, 2020-02-04:
IMesh works disabling this option but the log again return the error:
Feb  5 00:18:25 dnsmasq-script[363]: connect error: No such file or directory
Feb  5 00:18:25 dnsmasq-script[363]: [SEND_AMAS_NODE_EVENT:(4684)] ERROR connecting:No such file or directory.
Feb  5 00:18:25 dnsmasq-script[363]: connect error: No such file or directory
Feb  5 00:18:25 dnsmasq-script[363]: [SEND_AMAS_NODE_EVENT:(4684)] ERROR connecting:No such file or directory.
Feb  5 00:18:26 dnsmasq-script[363]: connect error: No such file or directory
Feb  5 00:18:26 dnsmasq-script[363]: [SEND_AMAS_NODE_EVENT:(4684)] ERROR connecting:No such file or directory.
Feb  5 00:18:34 rc_service: cfg_server 951:notify_rc restart_acsd

All the time

closed 2021-01-03 by RMerl