RMerl/asuswrt-merlin.ng #412: Update and Signature check fail V384.14_2 RT-AC3100

peepsnet, 2020-01-22:
Ver: 384.14_2
Router: RT-AC3100
Page: /Advanced_FirmwareUpgrade_Content.asp

The Signature Check  and Update check buttons fail:
"Signature update failed"
"Temporarily unable to get the latest firmware information. Please try again later."

I have tried multiple times over multiple days.

RMerl, 2020-01-23:
Trend Micro signature updates are done within closed source components, and talk directly with Asus's servers, so this is outside of my control.

closed 2020-01-23 by RMerl

peepsnet, 2020-01-23:
But the firmware update check fails also. Doesn't this check for this firmware?

RMerl, 2020-01-23:
If both fails, then something is misconfigured with your router.  I would start by checking your WAN DNS configuration.

peepsnet, 2020-01-23:
It worked with the previous version. It only stopped last week after I upgraded to 384.14_2
Although I am using OpenDNS it is not restricted. 

I have searched through all blocked DNS requests and I have ZERO blocked requests this week. ( i checked for updates yesterday and just before this post. )
I can look through the DNS logs on openDNS. What is the domain I am looking for?

peepsnet, 2020-01-23:
Was able to find this:

RANK | DOMAIN | REQUESTS
819 | fwupdate.asuswrt-merlin.net | 2
  | This domain resolved normally. You can block this domain or block similar domains .