RMerl/asuswrt-merlin.ng #415: Wrong router DNS resolved when OpenDNS IP filtering is enabled

mieszkomularczyk, 2020-01-26:
Test case:
RT-AC68 with 384.14_2, static external IP. External IP registered in a public domain like. router.mydomain.com, PC connected to local network in the router.

Scenario 1)
PC tries to resolve router.mydomain.com, and it gets 192.168.1.1 (correct), therefore, windows allows me to connect to \\router.mydomain.com\NAS for example.
Scenario 2)
I enable OpenDNS IP filtering. Then DNS query from my local PC connect to the router resolved to 212.x.y.z (public IP).
Windows reports DNS server as 192.168.1.1 when I do in cmd line : ipconfig /all, which means that all DNS traffic goes through DNS server on the router also when OpenDNS filtering is on.

Problem: no matter if IP filtering is on or off, DNS query for router.mydomain.com should always resolve to the NAT address. Could you add it in the code?

mieszkomularczyk, 2020-01-26:
I also checked /etc/hosts file on the router.  It contains entry for 192.168.1.1 router.mydomain.com, therefore even in IP filtering enabled, it should take precedence over OpenDNS

dave14305, 2020-01-27:
If you mean DNSFilter pointing to OpenDNS, that will bypass the router DNS completely, by design of iptables rules.

Why not set OpenDNS in WAN DNS and set DNS Filter to Router? You can have the best of both worlds.

mieszkomularczyk, 2020-01-27:
Hi @dave14305. I agree that you have a good 'workaround'. However I still believe that /etc/hosts file should not be completely ignored if one enables OpenDNS filtering in the GUI. Espeically in that case, when I am talking about automatically genereted /etc/hosts and I mean the entry pointing to the router itslef in the NAT address space (192.168.1.1).

dave14305, 2020-01-27:
Unfortunately, since the DNSFilter is actually performed by firewall rules, the DNS request is intercepted and forwarded to OpenDNS before it reaches the router’s DNS server. Even though you see the router IP as the DNS server on the PC, the firewall will redirect it. It is a trade off when using DNSFilter. See the last paragraph of the wiki page: https://github.com/RMerl/asuswrt-merlin/wiki/DNS-Filter

mieszkomularczyk, 2020-01-27:
You are right, it is clearly described as a "feature" not bug.

closed 2020-01-27 by mieszkomularczyk