RMerl/asuswrt-merlin.ng #417: cannot disable https via GUI

B-Art, 2020-02-04:
I am unable to "disable https access" because WAN access is still enabled...
(message occurs when I try to change change from BOTH to HTTP only.)
Solution is to reset the router, but I am missing the WAN access disable possibility.

Antonius-git, 2020-02-06:
Apologises for hijacking but there is an extension to this bug:
Cannot disable **http** via GUI.

This has been a problem for a while now. In the GUI, I set "Admin" > "System" > "Local Access Config" to **https only** but **http** access is still working (as if the **both** option were being chosen) even with altered port numbers.

This is confirmed with netstat which shows both instances are running. Manually killing the PID for the http instance works, leaving only https access on the given port.

dave14305, 2020-02-07:
On my router, with https only enabled, httpd is only listening on the loopback interface port 80, not the LAN interface.

Antonius-git, 2020-02-07:
With https only enabled, anyone on my LAN can access the router on http://192.168.0.1 or https://192.168.0.1:8443 until that httpd instance gets killed to make it truly **https only**, so it's definitely LAN facing which is a minor annoyance but it means the choice of 'https''http' or 'both' is not being honoured.

dave14305, 2020-02-07:
If you login to the router and run "nvram get http_enable" does it give 1 (https only)?

Antonius-git, 2020-02-07:
Indeed.
See also:

```
nvram show | grep http
apps_ipkg_server=http://nw-dlcdnet.asus.com/asusware/arm/stable
size: 65366 bytes (65706 left)
firmware_server=https://fwupdate.asuswrt-merlin.net
http_autologout=30
http_client=1
http_clientlist=
http_dut_redir=0
http_enable=1
http_id=[[redacted]]
http_lanport=80
http_passwd=[[redacted]]
http_username=[[redacted]]
httpd_die_reboot=
https_crt_cn=
https_crt_file=
https_crt_gen=0
https_crt_save=1
https_lanport=8443
last_httpd_handle_request=
last_httpd_handle_request_fromapp=
le_acme_auth=http
misc_http_x=0
misc_httpport_x=8080
misc_httpsport_x=8443
pushnotify_httplogin=1
webdav_http_port=8082
webdav_https_port=443
```
Bare in mind, I think I once tried setting **http_enable** to 0 and that blocked both http AND https access.

Antonius-git, 2020-02-08:
> On my router, with https only enabled, httpd is only listening on the loopback interface port 80, not the LAN interface.

To be fair, unlike OP these are not WAN facing, but:
(output after rebooting from new merlin firmware update)
```
netstat -peanut | grep http
tcp        0      0 127.0.0.1:80           0.0.0.0:*               LISTEN      1000/httpd
tcp        0      0 192.168.0.1:80         0.0.0.0:*               LISTEN      1000/httpd
tcp        0      0 127.0.0.1:8443         0.0.0.0:*               LISTEN      999/httpds
tcp        0      0 192.168.0.1:8443       0.0.0.0:*               LISTEN      999/httpds
```
and then in this case (PID varies of course)
``kill -9 1000``

The quick workaround is going to be a line of code on a startup script that finds the PID of 'httpd' and ends this instance so that https only is honoured. But it would be nice if the bug could get a look at since it seems like the **http only**/**https only**/**both** choice is just falling through to both for some reason.

delusion2019, 2020-02-23:
Another, better workaround is to install the Asus Router app on a mobile phone and choose in settings HTTPS only and than remove the app

Antonius-git, 2020-02-23:
> Another, better workaround is to install the Asus Router app on a mobile phone and choose in settings HTTPS only and than remove the app

That's a better workaround?

1) Not everyone has a smartphone even in 2020 capable of using apps - yes, how bizarre!
2) Does this method actually work properly without leaving http access available as well and vice versa (the entire point of this bug report)?
3) If it _does_ work by performing this workaround, why and can't that lead to an actual fix in the actual code on the actual hardware without resorting to other devices?
 
If anyone were to follow your workaround (assuming it is a workaround), everytime the router was rebooted (e.g. firmware upgrade), they'd have to reinstall and remove this app. At least with my suggestion it can be automated at startup.

commit 3a75772e (2020-03-13): webui: only forbid setting protocol to http-ony if in router mode (fixes #417)
commit c8432308 (2020-03-14): webui: only forbid setting protocol to http-ony if in router mode (fixes #417)
closed 2020-03-13 by RMerl