RMerl/asuswrt-merlin.ng #420: iperf3,wget,curl,ssh,scp not support ipv6

haohetao, 2020-02-08:
iperf3,wget,curl,ssh,scp command is timeout when host address is ipv6 address.

RMerl, 2020-02-18:
What firmware are you running?  Iperf3 is not part of my firmware.

haohetao, 2020-02-22:
iperf3 from entware-ng
The problem in the ip6tables
ip6tables deny all packet to INPUT Chain
I run 

> ip6tables -D INPUT -j DROP

to fix

RMerl, 2020-02-22:
Don't do that.  You are opening your whole router to the entire world by doing this.

haohetao, 2020-02-22:
@RMerl the ip6tables default deny all packet to router host(not foward),also can't configuration from web ui.
I disabled firewall for tested,not really.

stehpancar, 2020-03-02:
What would the most restrictive rule be to allow curl, wget and all the other to function with IPv6 hosts (and their -6 option)?
Would `ip6tables -I INPUT 1 -j ACCEPT --match state --state ESTABLISHED`, similar to existing iptables (ipv4) INPUT rule?

I also need curl (or wget) working for a custom Dynamic DNS update script to update IPv6 entries.

closed 2021-01-03 by RMerl