RMerl/asuswrt-merlin.ng #421: 384.15 DHCP start IP reset after submit LAN IP Apply button

wincorex, 2020-02-10:
LAN IP / Server DHCP / Start IP range - will reset to 192.168.2.2 from 192.168.2.100 when LAN IP / Hostname - change and Apply.

dave14305, 2020-02-11:
There should be a pop-up warning about this after you hit apply. Happened to me before because I didn't read the warning properly.

wincorex, 2020-02-12:
Oh, there is, so it's probably localization issue, because czech translation of this pop-up message does not mentioning it, or not in  understandable way at least.

nospam2k, 2020-02-17:
So, is this not a bug? While I understand that changing the router's IP address SHOULD reset the dhcp range, changing anything else (domain, gateway, etc.) should not.

wincorex, 2020-02-17:
Ok, it's not bug.
Close.

closed 2020-02-17 by wincorex