RMerl/asuswrt-merlin.ng #422: Tunnel 6in4 doesnt work when aimesh is active

aalmenar, 2020-02-10:
Hi and thanks for the effort of bringing this fine software to our hands.

Today i would like to report a strange issue im seeing.

Using version 384.11, 12, 13 14 and 15 configuring a tunnel 6in4 with tunnelbroker works fine, but if the router has another router connected via AiMesh, the tunnel no longer comes up.

The issue is really strange when having AiMesh active, since its possible to ping client and server side ipv6, but ipv6 traffic is not routed.

If AiMesh is deactivated it works!.

Using an AC68U router.

Please let me know if anything else is needed to help with this issue.

aalmenar, 2020-02-10:
IPv6 Routing Table: (Modified to not show the real ip values)

2001:470:XXXX:XXXX::1 dev v6tun0  metric 1  mtu 1480 advmss 1420 hoplimit 0
2001:470:XXXX:XXXX::/64 via :: dev v6tun0  proto kernel  metric 256  mtu 1480 advmss 1420 hoplimit 0
2600:XXXX:XXXX::/48 dev br0  proto kernel  metric 256  mtu 1500 advmss 1440 hoplimit 0
fe80::/64 dev eth0  proto kernel  metric 256  mtu 1500 advmss 1440 hoplimit 0
fe80::/64 dev eth1  proto kernel  metric 256  mtu 1500 advmss 1440 hoplimit 0
fe80::/64 dev eth2  proto kernel  metric 256  mtu 1500 advmss 1440 hoplimit 0
fe80::/64 dev vlan1  proto kernel  metric 256  mtu 1500 advmss 1440 hoplimit 0
fe80::/64 dev wl0.1  proto kernel  metric 256  mtu 1500 advmss 1440 hoplimit 0
fe80::/64 dev wl1.1  proto kernel  metric 256  mtu 1500 advmss 1440 hoplimit 0
fe80::/64 dev br0  proto kernel  metric 256  mtu 1500 advmss 1440 hoplimit 0
fe80::/64 dev ifb0  proto kernel  metric 256  mtu 1500 advmss 1440 hoplimit 0
fe80::/64 dev ifb1  proto kernel  metric 256  mtu 1500 advmss 1440 hoplimit 0
fe80::/64 dev dpsta  proto kernel  metric 256  mtu 1500 advmss 1440 hoplimit 0
fe80::/64 via :: dev v6tun0  proto kernel  metric 256  mtu 1480 advmss 1420 hoplimit 0
default via 2001:470:XXXX:XXXX::1 dev v6tun0  metric 1  mtu 1480 advmss 1420 hoplimit 0

aalmenar, 2020-04-23:
Anything that can be done to help on this issue?

asaddon, 2020-04-23:
If the issue is due to AiMesh then you have to contact Asus because it is closed source and Merlin can't do anything about it.

closed 2021-01-03 by RMerl