RMerl/asuswrt-merlin.ng #423: OpenVPN Config not going into effect until restart

mpratt14, 2020-02-12:
First off, not sure if this is by design or an actual issue.

If I change the config in any way while the OpenVPN client is active, for example, adding a route to custom config, and click apply, all it does is save it to the text file, but none of it goes into effect. This made me think for a while that something was wrong with my server, as the route I added wasn't functioning.

In OpenWRT for example, if any config change is applied, the process gets a hard restart (SIGTERM, then init). If your server is set up very well and has good internet, the clients can reconnect in a matter of 5 seconds, which in most cases is negligible downtime. If this is too extreme or difficult, maybe there can be a popup or text on the page saying something like "must be restarted manually for changes to go into effect". Also, having a restart button next to the On/Off toggle would be helpful for those who might be using the VPN to access the GUI

PS: Thanks for everything Merlin et al.

RMerl, 2020-03-13:
Are you referring to changes made to the server page, or the client page?

mpratt14, 2020-03-13:
I noticed this on the client page, but I assume it applies to both.

If I add or delete lines (like `route 192.168.x.0 255.255.255.0`)  in the "Custom Configuration" box at the bottom of the page, and click "Apply", the only thing it does is edit the actual config file. The OpenVPN process does not restart or reload that config file in any way. Changes only go into effect by toggling the client (or possibly server) process off, then on, at the top of the page, after applying changes. 

There's many ways to handle this, but if the functionality remains unchanged and the user doesn't know this, it is cause for a lot of confusion, where the assumption of a user is that clicking "Apply" would cause all changes to go into effect immediately (like it does, for example, in OpenWRT).

RMerl, 2020-03-14:
Unable to reproduce.  I connected my client, removed a line from my custom settings, clicked on Apply, and the client properly got restarted:

```
Mar 14 00:37:20 rc_service: httpd 1169:notify_rc restart_vpnclient1
Mar 14 00:37:20 ovpn-client1[11662]: event_wait : Interrupted system call (code=4)
Mar 14 00:37:20 ovpn-client1[11662]: SIGTERM received, sending exit notification to peer
Mar 14 00:37:21 ovpn-client1[11662]: vpnrouting.sh tun11 1500 1585 10.8.2.10 255.255.255.0 init
Mar 14 00:37:21 openvpn-routing: Configuring policy rules for client 1
Mar 14 00:37:21 ovpn-client1[11662]: Closing TUN/TAP interface
Mar 14 00:37:21 ovpn-client1[11662]: /sbin/ifconfig tun11 0.0.0.0
Mar 14 00:37:21 ovpn-client1[11662]: updown.sh tun11 1500 1585 10.8.2.10 255.255.255.0 init
Mar 14 00:37:21 rc_service: service 12057:notify_rc updateresolv
Mar 14 00:37:21 rc_service: waitting "restart_vpnclient1" via httpd ...
Mar 14 00:37:31 ovpn-client1[12164]: OpenVPN 2.4.8 arm-buildroot-linux-gnueabi [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD] built on Mar 11 2020
Mar 14 00:37:31 ovpn-client1[12164]: library versions: OpenSSL 1.1.1d  10 Sep 2019, LZO 2.08
```

Please post your system log following a click on Apply.

Also please report the firmware version and router model.

mpratt14, 2020-03-14:
Sorry, I should have been more detailed in first post...

RT-AC68P
384.14

When I removed a line and clicked Apply, time was 02:05, as you can see there's nothing at that time.

Should I be changing verbosity in the config for this or does it matter?

```
Mar 14 01:39:31 dnsmasq-dhcp[260]: DHCPSOLICIT(br0) 00:03:00:01:c8:d3:ff:7d:a2:e5 
Mar 14 01:41:31 dnsmasq-dhcp[260]: DHCPSOLICIT(br0) 00:03:00:01:c8:d3:ff:7d:a2:e5 
Mar 14 01:43:31 dnsmasq-dhcp[260]: DHCPSOLICIT(br0) 00:03:00:01:c8:d3:ff:7d:a2:e5 
Mar 14 01:44:28 ovpn-client2[11145]: VERIFY OK: depth=1, C=US, ST=xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
Mar 14 01:44:28 ovpn-client2[11145]: VERIFY KU OK
Mar 14 01:44:28 ovpn-client2[11145]: Validating certificate extended key usage
Mar 14 01:44:28 ovpn-client2[11145]: ++ Certificate has EKU (str) TLS Web Server Authentication, expects TLS Web Server Authentication
Mar 14 01:44:28 ovpn-client2[11145]: VERIFY EKU OK
Mar 14 01:44:28 ovpn-client2[11145]: VERIFY X509NAME OK: C=US, ST=xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
Mar 14 01:44:28 ovpn-client2[11145]: VERIFY OK: depth=0, C=US, ST=xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
Mar 14 01:44:29 ovpn-client2[11145]: Outgoing Data Channel: Cipher 'AES-256-GCM' initialized with 256 bit key
Mar 14 01:44:29 ovpn-client2[11145]: Incoming Data Channel: Cipher 'AES-256-GCM' initialized with 256 bit key
Mar 14 01:44:29 ovpn-client2[11145]: Control Channel: TLSv1.2, cipher TLSv1.2 ECDHE-RSA-AES256-GCM-SHA384, 2048 bit RSA
Mar 14 01:45:31 dnsmasq-dhcp[260]: DHCPSOLICIT(br0) 00:03:00:01:c8:d3:ff:7d:a2:e5 
Mar 14 01:45:54 dnsmasq-dhcp[260]: DHCPREQUEST(br0) 192.168.4.155 e8:2a:ea:75:f7:33 
Mar 14 01:45:54 dnsmasq-dhcp[260]: DHCPACK(br0) 192.168.4.155 e8:2a:ea:75:f7:33 ZB420
Mar 14 01:47:04 syslog: WLCEVENTD wlceventd_proc_event(386): wl0.1: Deauth_ind DC:56:E7:C4:FA:94, status: 0, reason: Deauthenticated because sending station is leaving (or has left) IBSS or ESS (3)
Mar 14 01:47:04 syslog: WLCEVENTD wlceventd_proc_event(420): wl0.1: Auth DC:56:E7:C4:FA:94, status: 0, reason: d11 RC reserved (0)
Mar 14 01:47:04 syslog: WLCEVENTD wlceventd_proc_event(449): wl0.1: Assoc DC:56:E7:C4:FA:94, status: 0, reason: d11 RC reserved (0)
Mar 14 01:47:05 dnsmasq-dhcp[260]: DHCPREQUEST(br0) 192.168.4.120 dc:56:e7:c4:fa:94 
Mar 14 01:47:05 dnsmasq-dhcp[260]: DHCPACK(br0) 192.168.4.120 dc:56:e7:c4:fa:94 iPhone
Mar 14 01:47:08 dnsmasq-dhcp[260]: DHCPREQUEST(br0) 192.168.4.120 dc:56:e7:c4:fa:94 
Mar 14 01:47:08 dnsmasq-dhcp[260]: DHCPACK(br0) 192.168.4.120 dc:56:e7:c4:fa:94 iPhone
Mar 14 01:47:31 dnsmasq-dhcp[260]: DHCPSOLICIT(br0) 00:03:00:01:c8:d3:ff:7d:a2:e5 
Mar 14 01:49:31 dnsmasq-dhcp[260]: DHCPSOLICIT(br0) 00:03:00:01:c8:d3:ff:7d:a2:e5 
Mar 14 01:51:31 dnsmasq-dhcp[260]: DHCPSOLICIT(br0) 00:03:00:01:c8:d3:ff:7d:a2:e5 
Mar 14 01:53:31 dnsmasq-dhcp[260]: DHCPSOLICIT(br0) 00:03:00:01:c8:d3:ff:7d:a2:e5 
Mar 14 01:55:31 dnsmasq-dhcp[260]: DHCPSOLICIT(br0) 00:03:00:01:c8:d3:ff:7d:a2:e5 
Mar 14 01:57:31 dnsmasq-dhcp[260]: DHCPSOLICIT(br0) 00:03:00:01:c8:d3:ff:7d:a2:e5 
Mar 14 01:59:31 dnsmasq-dhcp[260]: DHCPSOLICIT(br0) 00:03:00:01:c8:d3:ff:7d:a2:e5 
Mar 14 02:01:31 dnsmasq-dhcp[260]: DHCPSOLICIT(br0) 00:03:00:01:c8:d3:ff:7d:a2:e5 
Mar 14 02:05:31 dnsmasq-dhcp[260]: DHCPSOLICIT(br0) 00:03:00:01:c8:d3:ff:7d:a2:e5 
Mar 14 02:07:31 dnsmasq-dhcp[260]: DHCPSOLICIT(br0) 00:03:00:01:c8:d3:ff:7d:a2:e5 
Mar 14 02:09:31 dnsmasq-dhcp[260]: DHCPSOLICIT(br0) 00:03:00:01:c8:d3:ff:7d:a2:e5 
```

mpratt14, 2020-03-14:
From SSH output, the routes I deleted were 10.1.10.0 and 192.168.100.1 and after applying they are still in the file /tmp/etc/openvpn/client2/config.ovpn and still in the routing table.

```
# Custom Configuration
resolv-retry infinite
keepalive 10 120
persist-tun
persist-key
auth-nocache
remote-cert-tls server
route 192.168.10.0 255.255.255.0
route 192.168.100.0 255.255.255.0
route 192.168.101.0 255.255.255.0
route 10.1.10.0 255.255.255.0
420@RT-AC68P-1128:/tmp/etc/openvpn/client2# ip route list
73.xxx.0.1 dev eth0  proto kernel  scope link
192.168.100.0/24 via 10.8.0.1 dev tun12
192.168.101.0/24 via 10.8.0.1 dev tun12
192.168.4.0/24 dev br0  proto kernel  scope link  src 192.168.4.1
10.1.10.0/24 via 10.8.0.1 dev tun12
10.8.0.0/24 dev tun12  proto kernel  scope link  src 10.8.0.4
192.168.10.0/24 via 10.8.0.1 dev tun12
73.xxx.0.0/23 dev eth0  proto kernel  scope link  src 73.xxx.x.21
127.0.0.0/8 dev lo  scope link
default via 73.xxx.0.1 dev eth0
420@RT-AC68P-1128:/tmp/etc/openvpn/client2#
```

RMerl, 2020-03-16:
According to your log, no event was sent by the webpage to the web server.

I only tested it on the RT-AX88U, I will have to retest it on the RT-AC68U (different codebase).

RMerl, 2020-03-16:
Tested on an RT-AC68U, still unable to reproduce here.

mpratt14, 2020-03-16:
Chrome Console only shows 404 errors for resources from /images

Firefox Console shows this after hitting apply:

`Form contains a file input, but is missing method=POST and enctype=multipart/form-data on the form.  The file will not be sent.`
and it points to
`Advanced_OpenVPNClient_Content.asp:532:14`

It's a warning not an error and it disappears after page reloads.

Is there anything else I can try to show the problem better? Can I trigger it with curl somehow, or in SSH? Maybe whatever the web server is calling isn't working, like testing rc_service or rc_notify?

RMerl, 2020-03-16:
I checked the webpage code.  Whenever someone clicks on Apply, if the vpnclient2 process is found running, then it will issue a restart_vpnclient2 request.

Please confirm that the process is indeed running under its intended name.

```
ps w | grep vpnclient
```

mpratt14, 2020-03-16:
```
ASUSWRT-Merlin RT-AC68U 384.14-0 Sat Dec 14 00:39:28 UTC 2019
420@RT-AC68P-1128:/tmp/home/root# ps w | grep vpn
21351 420       4416 S    /etc/openvpn/vpnclient2 --cd /etc/openvpn/client2 --config config.ovpn
25114 420       1416 S    grep vpn
420@RT-AC68P-1128:/tmp/home/root#
```

mpratt14, 2020-03-16:
I got it to work, but I have no idea which browser I was using and whether I changed a line or not...
Will have to play with it more

```
Mar 15 20:51:41 dnsmasq[260]: using nameserver 8.8.8.8#53
Mar 15 20:51:41 dnsmasq[260]: using nameserver 64.6.64.6#53
Mar 15 20:51:41 dnsmasq[260]: using only locally-known addresses for domain projectkw
Mar 15 20:51:42 custom_script: Running /jffs/scripts/openvpn-event (args: tun12 1500 1552 10.8.0.4 255.255.255.0 init)
Mar 15 20:51:44 ovpn-client2[20927]: /sbin/route add -net 192.168.10.0 netmask 255.255.255.0 gw 10.8.0.1
Mar 15 20:51:45 ovpn-client2[20927]: /sbin/route add -net 192.168.101.0 netmask 255.255.255.0 gw 10.8.0.1
Mar 15 20:51:45 custom_script: Running /jffs/scripts/openvpn-event (args: tun12 1500 1552 10.8.0.4 )
Mar 15 20:51:45 ovpn-client2[20927]: Initialization Sequence Completed
Mar 15 20:52:08 rc_service: httpds 13919:notify_rc restart_vpnclient2
Mar 15 20:52:08 ovpn-client2[20927]: event_wait : Interrupted system call (code=4)
Mar 15 20:52:08 ovpn-client2[20927]: vpnrouting.sh tun12 1500 1552 10.8.0.4 255.255.255.0 init
Mar 15 20:52:09 openvpn-routing: Configuring policy rules for client 2
Mar 15 20:52:09 custom_script: Running /jffs/scripts/openvpn-event (args: tun12 1500 1552 10.8.0.4 255.255.255.0 init)
Mar 15 20:52:09 ovpn-client2[20927]: /sbin/route del -net 192.168.10.0 netmask 255.255.255.0
Mar 15 20:52:09 ovpn-client2[20927]: /sbin/route del -net 192.168.101.0 netmask 255.255.255.0
Mar 15 20:52:09 ovpn-client2[20927]: Closing TUN/TAP interface
Mar 15 20:52:09 ovpn-client2[20927]: /sbin/ifconfig tun12 0.0.0.0
Mar 15 20:52:09 lldpd[334]: removal request for address of 10.8.0.4%19, but no knowledge of it
Mar 15 20:52:09 ovpn-client2[20927]: updown.sh tun12 1500 1552 10.8.0.4 255.255.255.0 init
Mar 15 20:52:10 rc_service: service 21298:notify_rc updateresolv
Mar 15 20:52:10 rc_service: waitting "restart_vpnclient2" via httpds ...
Mar 15 20:52:19 kernel: Interface tap12 doesn't exist
Mar 15 20:52:19 kernel: Interface tun12 doesn't exist
Mar 15 20:52:19 dnsmasq[260]: read /etc/hosts - 10 addresses
```

closed 2021-01-03 by RMerl