RMerl/asuswrt-merlin.ng #424: User script - LAN/WiFi event

nullvalue1, 2020-02-13:
There is already a wan-event for things like connect/disconnect. There should be an ability to have a script execute when a client connects to the LAN and/or Wifi. This event would receive parameters about the connection.

Or does this capability already exist?

RMerl, 2020-02-14:
No possible, as I have no way of knowing when that occurs.

closed 2020-02-14 by RMerl

asaddon, 2020-02-14:
@RMerlin For the Wi-Fi event can't we use the Wireless Log as a base to determine the client connection/disconnection along with the MAC/IP address matching?

asaddon, 2020-02-14:
@RMerl ........
(Why can't I edit the comment on GH app) 🙄🙄

RMerl, 2020-02-14:
That would require constantly polling, and would only cover Wifi clients - too hackish and unreliable.  It's not an event then.