RMerl/asuswrt-merlin.ng #425: ntp not running after scheduled reboot

SebTardif, 2020-02-13:
Using ASUS RT-AC5300 with version 384.14_2 with two AiMesh Node the NTP is not run after scheduled reboot. This is a recent issue. Only change is likely about having added an extra AiMesh Node or having upgraded to 384.14_2. This is breaking AiProtection Parental Controls because the time is wrong. The UI know the time is wrong because under AIProtection it shows a warning that the time have not be synchronized/set. However, the settings is enabled. The if I go to the page about ntp, and just save without modification, it wake-up and do the synchronization. Doesn't make sense in many ways. The system log doesn't seem to have anything useful, doesn't include keyword "ntp".

I have attached a long log to this. You can see the 'reboot' when the time go backward.

SebTardif, 2020-02-13:
`
Feb 13 03:00:45 avahi-daemon[9084]: WARNING: No NSS support for mDNS detected, consider installing nss-mdns!
May  5 01:05:13 kernel: CPU: VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
May  5 01:05:13 kernel: Machine: Northstar Prototype
May  5 01:05:13 kernel: Ignoring unrecognised tag 0x00000000
May  5 01:05:13 kernel: Memory policy: ECC disabled, Data cache writealloc
May  5 01:05:13 kernel: Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 130048
May  5 01:05:13 kernel: Kernel command line: root=/dev/mtdblock2 console=ttyS0,115200 init=/sbin/preinit earlyprintk debug
May  5 01:05:13 kernel: Memory: 514896k/514896k available, 9392k reserved, 0K highmem
May  5 01:05:13 kernel: Virtual kernel memory layout:
May  5 01:05:13 kernel:     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
May  5 01:05:13 kernel:     fixmap  : 0xfff00000 - 0xfffe0000   ( 896 kB)
May  5 01:05:13 kernel:     DMA     : 0xf7e00000 - 0xffe00000   ( 128 MB)
May  5 01:05:13 kernel:     vmalloc : 0xa0800000 - 0xf0000000   (1272 MB)
May  5 01:05:13 kernel:     lowmem  : 0x80000000 - 0xa0000000   ( 512 MB)
May  5 01:05:13 kernel:     modules : 0x7f000000 - 0x80000000   (  16 MB)
May  5 01:05:13 kernel:       .init : 0x80008000 - 0x80050000   ( 288 kB)
May  5 01:05:13 kernel:       .text : 0x80050000 - 0x80410000   (3840 kB)
May  5 01:05:13 kernel:       .data : 0x8042a000 - 0x8044f440   ( 150 kB)
May  5 01:05:13 kernel: External imprecise Data abort at addr=0x0, fsr=0x1c06, pc=0x8000f9d8 lr=0x8000f9cc ignored.
May  5 01:05:13 kernel: Mount-cache hash table entries: 512
May  5 01:05:13 kernel: CPU1: Booted secondary processor
May  5 01:05:13 kernel: Found a AMD NAND flash:
May  5 01:05:13 kernel: Total size:  128MB
May  5 01:05:13 kernel: Block size:  128KB
May  5 01:05:13 kernel: Page Size:   2048B
May  5 01:05:13 kernel: OOB Size:    64B
May  5 01:05:13 kernel: Sector size: 512B
May  5 01:05:13 kernel: Spare size:  16B
May  5 01:05:13 kernel: ECC level:   8 (8-bit)
May  5 01:05:13 kernel: Device ID: 0x 1 0xf1 0x80 0x1d 0x 1 0xf1
May  5 01:05:13 kernel: ACP (Accelerator Coherence Port) enabled
May  5 01:05:13 kernel: bio: create slab <bio-0> at 0
May  5 01:05:13 kernel: PCI: no core
May  5 01:05:13 kernel: PCI: no core
May  5 01:05:13 kernel: PCI: Fixing up bus 0
May  5 01:05:13 kernel: PCI: Fixing up bus 0
May  5 01:05:13 kernel: PCIE: Doing ASMedia switch Init...Test Read = 20010002
May  5 01:05:13 kernel: PCIE 0001:01:0000: ASMedia UpPort mem_base 0x08000000, mem_limit 0x09ffffff
May  5 01:05:13 kernel: PCI: Fixing up bus 1
May  5 01:05:13 kernel: PCIE: Doing ASMedia switch Init...Test Read = 20010002
May  5 01:05:13 kernel: PCIE 0001:02:0018: ASMedia DownPort mem_base 0x0b000000, mem_limit 0x0cffffff
May  5 01:05:13 kernel: PCIE 0001:02:0018: ASMedia DownPort Link speed is GEN2
May  5 01:05:13 kernel: PCIE 0001:02:0018: ASMedia DownPort Link status 0x7012
May  5 01:05:13 kernel: PCIE: Doing ASMedia switch Init...Test Read = 20010002
May  5 01:05:13 kernel: PCIE 0001:02:0038: ASMedia DownPort mem_base 0x0e000000, mem_limit 0x0fffffff
May  5 01:05:13 kernel: PCIE 0001:02:0038: ASMedia DownPort Link speed is GEN2
May  5 01:05:13 kernel: PCIE 0001:02:0038: ASMedia DownPort Link status 0x7012
May  5 01:05:13 kernel: PCI: Fixing up bus 2
May  5 01:05:13 kernel: PCI: Fixing up bus 3
May  5 01:05:13 kernel: PCI: Fixing up bus 4
May  5 01:05:13 kernel: PCI: Fixing up bus 0
May  5 01:05:13 kernel: PCI: Fixing up bus 1
May  5 01:05:13 kernel: VFS: Disk quotas dquot_6.5.2
May  5 01:05:13 kernel: Dquot-cache hash table entries: 1024 (order 0, 4096 bytes)
May  5 01:05:13 kernel: PCIE: Doing ASMedia switch Init...Test Read = 20010002
May  5 01:05:13 kernel: PCIE: Doing ASMedia switch Init...Test Read = 20010002
May  5 01:05:13 kernel: pflash: found no supported devices
May  5 01:05:13 kernel: bcmsflash: found no supported devices
May  5 01:05:13 kernel: Boot partition size = 524288(0x80000)
May  5 01:05:13 kernel: lookup_nflash_rootfs_offset: offset = 0x200000
May  5 01:05:13 kernel: nflash: squash filesystem with lzma found at block 30
May  5 01:05:13 kernel: Creating 4 MTD partitions on "nflash":
May  5 01:05:13 kernel: 0x000000000000-0x000000080000 : "boot"
May  5 01:05:13 kernel: 0x000000080000-0x000000200000 : "nvram"
May  5 01:05:13 kernel: 0x000000200000-0x000004000000 : "linux"
May  5 01:05:13 kernel: 0x0000003c2ae4-0x000004000000 : "rootfs"
May  5 01:05:13 kernel: === PPTP init ===
May  5 01:05:13 kernel: Registering the dns_resolver key type
May  5 01:05:13 kernel: Spare area=64 eccbytes 56, ecc bytes located at:
[... 1203 lines cut ...]
May  5 03:43:05 syslog: WLCEVENTD wlceventd_proc_event(386): eth3: Deauth_ind EC:F4:51:80:74:81, status: 0, reason: Class 3 frame received from nonassociated station (7)
May  5 03:43:15 syslog: WLCEVENTD wlceventd_proc_event(386): eth3: Deauth_ind EC:F4:51:80:74:81, status: 0, reason: Class 3 frame received from nonassociated station (7)
May  5 03:43:25 syslog: WLCEVENTD wlceventd_proc_event(386): eth3: Deauth_ind EC:F4:51:80:74:81, status: 0, reason: Class 3 frame received from nonassociated station (7)
May  5 03:43:35 syslog: WLCEVENTD wlceventd_proc_event(386): eth3: Deauth_ind EC:F4:51:80:74:81, status: 0, reason: Class 3 frame received from nonassociated station (7)
May  5 03:43:45 syslog: WLCEVENTD wlceventd_proc_event(386): eth3: Deauth_ind EC:F4:51:80:74:81, status: 0, reason: Class 3 frame received from nonassociated station (7)
[... cut ...]
May  5 03:46:38 syslog: WLCEVENTD wlceventd_proc_event(386): eth3: Deauth_ind EC:F4:51:80:74:81, status: 0, reason: Class 3 frame received from nonassociated station (7)
May  5 03:46:48 syslog: WLCEVENTD wlceventd_proc_event(386): eth3: Deauth_ind EC:F4:51:80:74:81, status: 0, reason: Class 3 frame received from nonassociated station (7)
May  5 03:46:58 syslog: WLCEVENTD wlceventd_proc_event(386): eth3: Deauth_ind EC:F4:51:80:74:81, status: 0, reason: Class 3 frame received from nonassociated station (7)
May  5 03:47:08 syslog: WLCEVENTD wlceventd_proc_event(386): eth3: Deauth_i

RMerl, 2020-02-18:
Double check your DNS configuration, this is the most common cause of NTP failing to synchronize at boot time.

closed 2020-03-13 by RMerl