RMerl/asuswrt-merlin.ng #426: http://router (hostname=router, http port=80) will not accept login on refreshes page

nospam2k, 2020-02-14:
I can't login using the hostname "router" when the http port is 80.

What does work:
1. Change the http (not https) port to anything else.
2. Change the hostname (in LAN) to anything else. (all other hostname, domain entries are blank)
3. Enable redirect to router.asus.com

Router RT-AC68U from reflash of TM-AC1900.

RMerl, 2020-02-14:
The issue tracker is only for development purposes.  If you need technical support or have a question, please use the support forums at https://www.snbforums.com/forums/asuswrt-official.51/ instead.

closed 2020-02-14 by RMerl

nospam2k, 2020-02-14:
I'm a little confused. Is this not a development issue? I'm reporting a bug. I'm not looking for support. I am also a developer. This is easy to reproduce. I have also posted on sbnforums.

nospam2k, 2020-02-16:
I know you closed this without investigating but this is a bug. Here is what I've found so far:

If I wget using http://router and do another wget using http://router. I get the following  in Main_Login.asp returned:
```
var header_info = [{ "host": "router\r", "current_page": "Main_Login.asp", "protocol": "http", "port": 80 }][0];
var header_info = [{ "host": "router.", "current_page": "Main_Login.asp", "protocol": "http", "port":
```
Notice that router returns router\r as it's hostname but router. returns the same. I'm not sure why it does this but this is what I've found so far.