RMerl/asuswrt-merlin.ng #427: [AC68U] 8To HDD USB 3.0 SMB sharing not working

mime29, 2020-02-17:
I use an external HDD Seagate steb8000301 usb3.0
https://www.amazon.co.jp/dp/B07TZXR741/ref=cm_sw_r_cp_tai_PyPsEbTA4RF1Q

my goal: share its content with my apple tv 4k through SMB to read my digitalized blurays (I use Infuse on the apple tv, I also used vlc successfully)

problem: I plug the hdd on the usb 3.0 port (configured as a 3.0 port as well) and when I try to access its content and folders through the asus merlin interface SMB page by tapping on the hdd name, the waiting indicator keeps rotating for many minutes ( never waited more than 5 minutes ).
I also can't access the content from the apple tv softwares I referred earlier.

note: When I change the USB 3.0 port configuration to USB 2.0 then everything works fine except that the bandwidth is not enough to stream my digitalized bluray.
I can browse the folders in the SMB configuration panel of the asus merlin admin interface and I can also browse files kn vlc and infuse.

What could be the problem here?
Is it possible that 8To disks are not supported when plugged using usb3 but supported when using usb 2.0?

RMerl, 2020-02-18:
The issue tracker is only for development purposes.  If you need technical support or have a question, please use the support forums at https://www.snbforums.com/forums/asuswrt-official.51/ instead.

closed 2020-02-18 by RMerl

mime29, 2020-02-18:
I understand your comment but It sounds like a bug to me.
USB2 or USB3 configuration shouldn't trigger any problem to browser folders in the Asus Merlin SMB sharing screen.

mime29, 2020-02-18:
Please reconsider.
If you want I can change the description of the bug but I honestly though that it was enough detailed information to spot a real problem here.

RMerl, 2020-02-18:
As I said, post on the support forums, so the community can go through it with you.  The most likely reason for the behavior you describe is a compatibility or interference issue with the disk in question.  You haven't identified a confirmed bug, you just mentioned an issue with your setup, which requires troubleshooting first.

mime29, 2020-02-18:
> As I said, post on the support forums, so the community can go through it with you. The most likely reason for the behavior you describe is a compatibility or interference issue with the disk in [...]

Got it, I'll do my best to get more information about it and come back to you if necessary.
Thanks