RMerl/asuswrt-merlin.ng #428: DHCP start and rang IP resets incorrectly on changes to hostname or domain name

nospam2k, 2020-02-17:
While I understand that changing the router's IP address SHOULD reset the dhcp range, changing anything else (domain, gateway, etc.) doesn't need to.

Also, the range defaults to 192.168.0.2 - 192.168.0.254 even if the router ip is 192.168.0.254

wincorex, 2020-02-17:
It's probably not a bug,.. but i changed ONLY hostname, not IP, that's why, i was confused, that DHCP range did had reset, even if IP stayed same.

nospam2k, 2020-02-17:
I do not agree. I am not aware of other routers that do this. It is not necessary to modify the ipaddress range in dnsmasq when the router ip hasn't changed and requires having to change the config 2x every time you make a change to the router name or domain.

RMerl, 2020-02-18:
Tested, unable to reproduce.  I changed the domain, applied, and the DHCP range wasn't changed for me.

Are you sure it wasn`t your browser automatically filling those fields?

RMerl, 2020-02-18:
> Also, the range defaults to 192.168.0.2 - 192.168.0.254 even if the router ip is 192.168.0.254

This is a deliberate design decision from Asus.  If the router's IP is within your scope, then it will display a (poorly descriptive) warning at the top showing your router's IP address in yellow.

RMerl, 2020-02-18:
Warning message reworded to be more descriptive - 4d9cd0ec39ffe61fb1fbd098241ea14fdc609ea1

nospam2k, 2020-02-18:
I'm out of town this week, but I will investigate further. I have had this happen on several devices so I don't think it's the browser.

nospam2k, 2020-02-23:
> Tested, unable to reproduce. I changed the domain, applied, and the DHCP range wasn't changed for me.
> 
> Are you sure it wasn`t your browser automatically filling those fields?

Just to confirm, try testing setting the router to 192.168.0.254, then set the dhcp range to 192.168.0.30-192.168.0.49. If I do this and then ONLY change the router hostname, the dhcp range is set to 192.168.0.2-192.168.0.254 after the changes are saved. I've done this from several browsers including mobile and all produce the same results.

RMerl, 2020-02-23:
I can't see it if I change the domain, but I get it on a hostname change (and the page even shows an alert warning about IP change requiring a scope adjustment).  The odd thing however is that if afterward I change the hostname a second time, it will not complain and cause a scope reset.  So, the check must be conditional to the existing scope as well.  This will require further investigation.

commit 5c3c0d10 (2020-03-13): webui: no longer offer to reset DHCP scope when changing LAN hostname (fixes #428)
commit ee5c42fd (2020-03-14): webui: no longer offer to reset DHCP scope when changing LAN hostname (fixes #428)
closed 2020-03-13 by RMerl