RMerl/asuswrt-merlin.ng #437: Reboots & OpenVPN client & routing / firewall defect

combivanCoder, 2020-02-24:
Hi - 

I'm seeing the same problem and I haven't seen it noted elsewhere. Generally though this firmware is a massive step forward from the stock Asus firmware, and resolves all the security issues I've raised with Asus about their firmware (esp. on the GT boxes). Great work - thanks so much for all your efforts.

This is an odd situation and I'm hoping a little information can help identify the cause of the problem.

**High-LevelConfiguration in this scenario**
- Acting as a router, with a wired edge router dealing with the ISP connectivity
- DNS server off-router (dedicated server on network)
- DHCP server off-router (dedicated server on network)
- NTP server off-router (dedicated server on network)
- OpenVPN configured on one connection; default is VPN, exceptions added for specific machines to *not* use VPN; killswitch option on no VPN enabled
- OpenVPN uses ProtonVPN which involves certs within the configuration
- USB FS used for traffic history, haven't configured it for anything else yet

**Behaviour**
1. BAU is fine. Working as expected in most scenarios
2. On VPN fail the openvpn client doesn't always attempt reconnect, but usually does
3. On reboot the VPN client often fails with "AUTH_FAILED" error codes. When logging into the UX and hitting the "Apply" button on the VPN Client page OR `service restart_vpnclient1` get things moving again
4. Scheduled reboot always fails to route properly. When the VPN does connect properly only the router can ping out over the VPN connection. No other machine on the LAN connected to the router can get traffic out.

**Workarounds**
1. Logging into UX, navigating to VPN pages -> VPN Client and simply hitting the `Apply` button
2. Disabling scheduled reboot except on a couple of days when I know I'll be up before everyone else. Use these days to try and diagnose the fault / build workarounds
3. Cron job which pings over the VPN tunnel - if no ping, it restarts vpnclient no. 1. Runs every few minutes from around the time of the scheduled reboot for a couple of hours ("just in case").
4. WIP: Looking for a way to simulate the `Apply` button code invoked from a cron job atm

**General Notes**
It looks like the newly created workaround #3 is only solving half the problem. It's ensuring the VPN connection is connected but it doesn't appear to be unjamming whatever is causing the routing / firewall issue.

I'm wondering if this is perhaps rooted in the VPN configuration and maybe a race condition on boot? I'm guessing that the VPN configuration has to be stored somewhere, and the certs etc. needed for auth need to be available within the right place in the boot sequence - could this possibly be because the vpn client is trying to start before the config FS is available?

When I hit the `Apply` button on the VPN Client page I've noticed that the only change in iptables rules is that this rule:

```bash
Chain INPUT (policy ACCEPT 0 packets, 0 bytes)
 pkts bytes target     prot opt in     out     source               destination         
    0     0 ACCEPT     2    --  eth0   *       0.0.0.0/0            0.0.0.0/0        
```

...Disappears post-Apply. At that point LAN clients are able to route traffic happily.

janstrube, 2020-02-28:
Hi combivanCoder,

a number of people have experienced reliability issues with starting OpenVPN after a router reboot.  To resolve this issue you can delay the startup of the open vpn with a script. Create a script named 'init-start' in folder /jffs/scripts/

cat > init-start
#!/bin/sh
sleep 100
service start_vpnserver1
Press CTRL+D

Set the file attributes with chmod:
chmod a+rx /jffs/scripts/*

For more info on user scripts in general see here:
https://github.com/RMerl/asuswrt-merlin.ng/wiki/User-scripts

Cheers,
 Jan

combivanCoder, 2020-03-01:
> 
> a number of people have experienced reliability issues with starting OpenVPN after a router reboot. To resolve this issue you can delay the startup of the open vpn with a script. Create a script named 'init-start' in folder /jffs/scripts/
> 

Thanks @janstrube - good idea. The workaround seems to be effective.

As this is client (not server) related I've modified the script very slightly, but this seems to work for me:

```bash
#!/bin/sh

# copy the crontab and assocatiated scripts to /var/spool/cron/crontabs/
cp routername-crons.sh /var/spool/cron/crontabs/<admin username>
ln -s /var/spool/cron/crontabs/<admin username> /var/spool/cron/crontabs/admin

# on reboot, delay for 80s, then restart the vpn client(s)
sleep 80
service restart_vpnclient1
# ...repeat for all clients or servers
```

I've not been able to get the `@reboot cron` working yet, can't see any error messages so assume it's just not supported on this firmware? I would have liked to make this a bit neater with a:

```bash
@reboot logger "Workaround" "Sleeping 80s...then restarting the vpnclient..." && sleep 80 && service restart_vpnclient1
```

I've also had some conversations with Asus - Good exchange so far and they've been able to replicate what appears to be a race condition on the GT-series firmware affecting "VPN Fusion". Suspect something similar in the boot ordering is affecting the stock RT-AX88U firmware by the looks of it, and spilling over into Merlin.

Will monitor for a week and see if this workaround deals with the intermittent drops too.

combivanCoder, 2020-03-07:
Seems to be much more stable. Thanks!

closed 2020-03-07 by combivanCoder