RMerl/asuswrt-merlin.ng #439: I can't use deluge with deluge thin windows client.

Drigoo, 2020-02-27:
I am using deluge for entware in a version of Padavan, but I would like to have control by the deluge thin client to be able to use the yarss2 plugin (because it is possible to configure it only by accessing this way), but I cannot connect it to the thin client. Is there any extra configuration to be done, like adding a different user? I appreciate the help.

asaddon, 2020-02-27:
What this issue have to do with RMerl or Asuswrt-Merlin firmware? Ask on SNB support forums or contact Entware/Deluge/Padavan support.

RMerl, 2020-02-27:
This is totally unrelated to my firmware.  Please contact either the Padawan developers or the Entware developers.

closed 2020-02-27 by RMerl

Drigoo, 2020-02-27:
> O que esse problema tem a ver com o firmware RMerl ou Asuswrt-Merlin? Pergunte nos fóruns de suporte do SNB ou entre em contato com o suporte da Entware / Deluge / Padavan.

As far as I know, the Entwar repository is the same, so I thought I could clear up the doubt here. If you don't want to help, I understand, so please let it be boring and go about your business.

asaddon, 2020-02-27:
Seriously?? It's like asking for help for a Motor Bike from a Car mechanic just because they both use Petrol lol.