RMerl/asuswrt-merlin.ng #440: dnsmasq.postconf executed before '/tmp/resolv.dnsmasq' and '/tmp/resolv.conf' created.

kissglay, 2020-03-05:
After reboot, when the dnsmasq.postconf gets executed, '/tmp/resolv.dnsmasq' and '/tmp/resolv.conf' are not yet created. So there's no way to get what's in these files.
If the files are there and one modified something affecting it(e.g. if using DOT), dnsmasq.postconf gets the old stuff just before the files get updated.

paldier, 2020-03-05:
wanduck create and update resolv.conf,not dnsmasq
start dnsmasq->start wan->start wanduck

kissglay, 2020-03-05:
resolv.conf isn't of much concern.
What about "resolv.dnsmasq"? Does wanduck also create this one?

paldier, 2020-03-05:
yes

dave14305, 2020-03-05:
Try using service-event-end script after a dnsmasq action.

kissglay, 2020-03-06:
I didn't see any service-event-end entry with dnsmasq as args unless manually restart it. And will this work (i.e. are there any race conditions) if '/tmp/resolv.dnsmasq' is created or modified afterward?

dave14305, 2020-03-06:
Maybe if you explain what you want to accomplish I can give a better answer.

kissglay, 2020-03-06:
I want to add other 'server=' directions. I'm now pointing 'servers-file=' to another file and dynamically change the content as 'resolv.dnsmasq' using script. But I must keep tracking this file in the future release to see if something changed in it and modify the script accordingly. If I could just read this file and append my stuff, it'd be a lot easier.

Or, will these 'server=' direction work if directly appended to dnsmasq.conf?
Thx

dave14305, 2020-03-06:
You can remove the servers-file line via dnsmasq.postconf and append you’re desired server= lines in the same script. Then resolv.dnsmasq is irrelevant. Many dns solutions (Unbound, NextDNS, DNSCrypt-Proxy) do a similar method to redirect dnsmasq to upstream server different than the WAN DNS servers.

pc_delete "servers-file" "$1"
pc_append "server=127.0.0.1#53535" "$1"

kissglay, 2020-03-06:
What if I want to retain what's in resolv.dnsmasq? Because if you disable DoT or something, this file would be changed, containing the DNS specified in the WAN Page. Now I'm doing this kind of thing in the postconf.
What I need from 'server=' is to add some specific site to bypass DoT. (i.e. 'server=/xx.com/yy.com/1.1.1.1').
Will these directions work if I append them in dnsmasq.conf, providing 'servers-file=' is still there?
Thx

dave14305, 2020-03-06:
Yes, people also do that (usually with dnsmasq.conf.add) to direct ntp hostnames to a specific dns server to avoid problems at boot time if using dnssec or DoT.

dave14305, 2020-03-06:
You can get much more help on the forum: https://www.snbforums.com/forums/asuswrt-merlin.42/

kissglay, 2020-03-06:
Ok. then.
Thank you

closed 2020-03-06 by kissglay