RMerl/asuswrt-merlin.ng #444: miniupnpd with double NAT

xmentos, 2020-03-11:
Hi, I have a double NAT situation cause my ISP modem+router can't be set to bridge mode; I've activated it's DMZ however on the IP of my Asus RT-AC86U (384.15 firmware). Now I can see on my devices and in the router logs that miniupnp doesn't support this:
`Mar 11 21:48:22 miniupnpd[13807]: SendNATPMPPublicAddressChangeNotification: cannot get public IP address, stopping`

@RMerl , I saw your issue at miniupnp (https://github.com/miniupnp/miniupnp/issues/333), has there been any outcome with some additional settings in Merlin? Possibly by configuring a Stun service? I don't find it under the WAN settings.

RMerl, 2020-03-13:
The whole stun idea they suggest makes no sense, since the router's upnp daemon is responsible for forwarding its own WAN interface, not whichever fronting router's WAN interface it has.  It's been two years so I don't remember all the details, but as far as I recall, I simply reverted their change that broke double NAT support, and never looked back at it since it resumed the original working behaviour.

If it's currently broken then it most likely is a new issue that was introduced afterward.

RMerl, 2020-03-13:
Looks like it's another check for private IPs added with commit 1861d810581 .  I will have to see if it can safely be reverted.

RMerl, 2020-03-13:
This check was added by upstream commit cce19781e67364d36a9068a42d5275836ee88c89 (miniupnpd: Add validation that public ip address is not reserved and is really public).  Should in theory be safe to revert.

xmentos, 2020-03-13:
Getting rid of that check works for me. I don't know the codebase but it's using buildroot right? Isn't it cleaner to use the original upnp sources and apply a patch instead of modifying directly the sources?

BTW upnp stopped working for me after flashing the stock asus-wrt to merlin, I guess asus-wrt is stuck on an old miniupnpd version without that check. Thanks for your swift action!

RMerl, 2020-03-16:
>  Isn't it cleaner to use the original upnp sources and apply a patch instead of modifying directly the sources?

Depends on your workflow.  Personally I don't like that, because it makes it harder to directly analyze the code since you are not able to visualize the exact code as it's going to be compiled, or analyzing how patches may affect one another.  So for development purposes, it's better to have the code in the final form it will be at compile time.

xmentos, 2020-03-17:
OK I understand, thanks! I'll try to build it, is there some documentation on building and dependencies? Simply run tools/build-all?

xmentos, 2020-04-01:
@RMerl is this part of the  384.13_5 beta release? I'm willing to try and report back.

RMerl, 2020-04-01:
Yes, the fix was backported to 384.13_5.

xmentos, 2020-04-06:
Confirmed working with firmware 384.16 on RT-AC86U, thanks a lot! Closing issue.

closed 2020-04-06 by xmentos