RMerl/asuswrt-merlin.ng #446: vpnclient2-5 connect but network clients are blocked

markster101, 2020-03-14:
Only vpnclient1 can be used and is working. When manually switching to vpnclient2,3,4 or 5 vpn client will connect but all network clients browsers are not connecting.

1. I did a full reset rt86u and configured router manually. Configured WAN page DoT and DNSSEC
2. DNS filter set to Router
3. At that point all clients can access internet
4. Add vpnclient1 (NordVPN) - DNS Disable, Routing Policy set 192.168.50.193/26
5. Tested vpn connects and network clients get VPN connection. At this point jffs is disabled.
6. Configured vpnclient2. DNS Disable and the same Policy rules
7. Turn off vpnclient1
8. Turn on vpnclient2. VPN status connected and visible IP address
9. At this point I am unable to access any we url. Dig command work from router.
10. Turn vpnclient2 off
11. Turn vpnclient1 on. Everything works

If I configure vpnclient1 and vpnclient2 for the same VPN server it still does not work.

RMerl, 2020-03-16:
Works for me.

If you have the client1 instance configured for blocking Internet access when tunnel goes down, then try disabling that feature.

markster101, 2020-03-16:
Thanks Eric

That indeed was the fix. Is there a maunal command I can use to enable that from a script after I switch vpn clients?.
I have a script that performs the actual switching between vpn clients and I can incormporate enable|diable Internet blocking flag.

Mark

RMerl, 2020-03-16:
I would recommend enabling/disabling the killswitch, then forcing a refresh of the routing table.

```
nvram set vpn_client1_enforce=0
service start_vpnrouting1
```

markster101, 2020-03-16:
Thank you. That is what I have been looking for.

markster101, 2020-03-16:
Eric,

Sorry to bather you. I manuall switched to vpnclient2 with GUI (kill switch is off), then SSH to the router and executed the commands. But once I execute below code it behavious the same. No network clients are able to access internet. Somthing else must be going on. If I revert and do vram set vpn_client2_enforce=0 and perform service start_vpnrouting2 it works again. So to me it looks as if Kill Switch cannot be used.

nvram set vpn_client2_enforce=1 //  Enable Kill Switch
service start_vpnrouting2 // restart vpn client

Mark

RMerl, 2020-03-16:
After switching to client2, you`ll want to disable the killswitch on client1, not 2.

markster101, 2020-03-16:
Eric,

My starting point for the POC is both vpnclient1 and vpnclient2 Kill switch is OFF and performing the POC with GUI only. Vpnclient1 is active.

1. Turn off vpnclient1
2. Turn on vpnclient2. Connection to internet is working.
3. Activate vpnclient2 Kill Switch. Connection is lost.
4. Deactivate vpnclient2 Kill Switch. Conneciton is back.

Perfromed the same with your suggested code and the effect is the same. Something is with the Kill Switch that is preventing the connection.

These are my vpn policy rules for both vpnclient1 and vpnclient2.
Mark

closed 2021-01-03 by RMerl