RMerl/asuswrt-merlin.ng #448: /jffs/scripts/service-start is missing after a refresh update to 384.15 use AMPM.

zw963, 2020-03-15:
as i describe on title, on older merlin system,  after init entware use `entware-setup.sh`, we can always see a script, named /jffs/scripts/service-start, when run it, which always try to start all services in /opt/etc/init.d

But, after use this new release, after init entware use new AMTM script, only generate two script,
`post-mount`,  `services-start`, where is the famous "service-start" go?

Thank you.

RMerl, 2020-03-16:
Amtm moved the unslung command to a post-mount script instead of services-start, as it is a more appropriate location.  Services-start might run too early, causing it to delay starting the Entware services, or flat out miss them, while post-mount will run after the required USB disk has been successfully mounted.

RMerl, 2020-03-16:
Also, there never was a service-start in the firmware, it always was services-start.

wengych, 2020-03-16:
I have same issue, and after reboot the router, I can not use opkg anymore without /opt link .

zw963, 2020-03-17:
> Also, there never was a service-start in the firmware, it always was services-start.

Sorry, i means `/jffs/scripts/services-start`.

Anyway, for those don't want broken older script guys, following is original file

```sh
#!/bin/sh

RC='/opt/etc/init.d/rc.unslung'

i=30
until [ -x "$RC" ] ; do
  i=$(($i-1))
  if [ "$i" -lt 1 ] ; then
    logger "Could not start Entware"
    exit
  fi
  sleep 1
done
$RC start
```

closed 2020-03-17 by zw963