RMerl/asuswrt-merlin.ng #451: http redirector for public cname to lan hostname or cname

goranschwarz, 2020-03-18:
This is more a question and/or a feature request
*(if this already exists, which I haven’t discovered yet, please let me know)*

I would like a feature in Merlin that redirects public *cnode* names to lan *hostname*

My public dns/ddns name is `acme.com`, but then I also have *cnames*  `web1.acme.com`, `web2.acme.com`, `web3.acme.com`etc...
I simply want to direct:
* `web1.acme.com` -->> `host1.home:80` (or an IP address)
* `web2.acme.com` -->> `host1.home:8080` 
* `web3.acme.com` -->> `host3.home:8080` 

So this is more or less like current *port forwarding* but on *cname* level instead (and only on HTTP traffic)!

I know that this isn’t possible on a tcp packet level (at least to me knowledge, because the cname/hostname isn’t part of the tcp packet, only ip addresses are)   
So for SSH (port 22) or similar I have to use router/merlin `Port Forwarding`   
But at a HTTP level it should be possible (like the `VirtualHost` in Apache)   
For example:
```
<VirtualHost *:80>
  ServerName etherpad.acme.com
  ProxyPass / http://localhost:9001/
</VirtualHost>
```
Ok, I can do the above on any Linux box on the LAN, but it would be nice to have it at the router level   
It would simply make life easier :)

Syst3mSh0ck, 2020-03-28:
I don't know if this is possible with getdns/stubby but it would be nice to have.

RMerl, 2020-03-29:
DNS and HTTP are two totally separate things.  If you need a redirection at the http level, then it must be done by an http proxy or an http server - a DNS server cannot do that for you, as it won't rewrite the HTTP request to the destination hostname.

closed 2020-03-29 by RMerl

goranschwarz, 2020-03-29:
@RMerl 
Well that was the feature request: **Can we implement a `http proxy/re-director` in Merlin**   

It would be nice with a *simple* UI directly in Merlin to redirect (like in my example with the Apache `VirtualHost` but much simpler with a UI)   
I mean Merlin already have a Web Server... which may be able to handle the redirection as well :)

Right now I have an *extra* machine just for that purpose, and it would be nice to have that embedded in the Merlin router...

RMerl, 2020-03-29:
Not possible, port 80 is already used by the firmware itself.

goranschwarz, 2020-03-30:
So any feature requests regarding the HTTP/HTTPS should go to the ASUS firmware team...   
And later on that's imported into the Merlin firmware?   
(or who has control over the HTTP/HTTPS code)