RMerl/asuswrt-merlin.ng #452: RT-AC86U crashing intermittently

ryaniskira, 2020-03-19:
**Device:** RT-AC86U
**Version:** 384.15 & 384.16 Beta 1

Since about 12 or so hours ago, my router keeps crashing completely. No life out of it at all except the activity lights on the front. I can't access the web ui, I can't SSH into it, hell it doesn't even respond to ping requests, and of course this kills internet access. This is the only bit of log that seems to stand out and this log spam happened around the time of one of the crashes:

```
Mar 18 13:34:16 kernel: FPM Pool 0: pool full
Mar 18 13:34:16 kernel: FPM Pool 0: pool full
Mar 18 13:34:16 kernel: FPM Pool 0: pool full
Mar 18 13:34:16 kernel: FPM Pool 0: pool full
Mar 18 13:34:16 kernel: FPM Pool 0: pool full
Mar 18 13:34:16 kernel: FPM Pool 0: pool full
Mar 18 13:34:16 kernel: FPM Pool 0: pool full
Mar 18 13:34:16 kernel: FPM Pool 0: pool full
Mar 18 13:34:16 kernel: FPM Pool 0: pool full
Mar 18 13:34:16 kernel: FPM Pool 0: pool full
Mar 18 13:34:16 kernel: FPM Pool 0: pool full
Mar 18 13:34:16 kernel: FPM Pool 0: pool full
Mar 18 13:34:16 kernel: FPM Pool 0: pool full
Mar 18 13:34:16 kernel: FPM Pool 0: pool full
Mar 18 13:34:16 kernel: FPM Pool 0: pool full
Mar 18 13:34:16 kernel: FPM Pool 0: pool full
Mar 18 13:34:16 kernel: FPM Pool 0: pool full
Mar 18 13:34:16 kernel: FPM Pool 0: pool full
Mar 18 13:34:16 kernel: FPM Pool 0: pool full
Mar 18 13:34:16 kernel: FPM Pool 0: pool full
Mar 18 13:34:16 kernel: FPM Pool 0: pool full
Mar 18 13:34:16 kernel: FPM Pool 0: pool full
Mar 18 13:34:16 kernel: FPM Pool 0: pool full
```

Syst3mSh0ck, 2020-03-28:
Overflow in the kernel somewhere. I would recommend a full Factory Reset.

ryaniskira, 2020-04-06:
@Syst3mSh0ck I've tried that a few times, even applied the beta firmware in the hopes it was a bug and a fix for it was out.

xmentos, 2020-04-07:
Hi @ryaniskira, I have a RT-AC86U as well, have a look below for memory comparison. I would also check `dmesg` for any out of the ordinary messages, my dmesg here (384.16):
[dmesg-rtac86u.txt](https://github.com/RMerl/asuswrt-merlin.ng/files/4442449/dmesg-rtac86u.txt)
.
`admin@RT-AC86U-2AC8:/tmp/home/root# cat /proc/meminfo 
MemTotal:         440420 kB
MemFree:          228040 kB
MemAvailable:     215864 kB
Buffers:               0 kB
Cached:            25380 kB
SwapCached:            0 kB
Active:            41452 kB
Inactive:           8056 kB
Active(anon):      24948 kB
Inactive(anon):      892 kB
Active(file):      16504 kB
Inactive(file):     7164 kB
Unevictable:           0 kB
Mlocked:               0 kB
SwapTotal:             0 kB
SwapFree:              0 kB
Dirty:                 0 kB
Writeback:             0 kB
AnonPages:         24076 kB
Mapped:            12264 kB
Shmem:              1712 kB
Slab:             139600 kB
SReclaimable:       3172 kB
SUnreclaim:       136428 kB
KernelStack:        1952 kB
PageTables:         1584 kB
NFS_Unstable:          0 kB
Bounce:                0 kB
WritebackTmp:          0 kB
CommitLimit:      220208 kB
Committed_AS:     102776 kB
VmallocTotal:   260046784 kB
VmallocUsed:       28300 kB
VmallocChunk:   260007576 kB
`

kissglay, 2020-04-07:
If it crashes even after Factory Reset, it's definitely a hardware failure, having nothing to do with firmware.
I had bought a brand new AX88U and it crashed intermittently right out of the box. I struggle with it for several days with numerous tries but not working. So I RMA it and got a new one, problem solved.
So don't waste time digging some remedy if a Factory Reset doesn't help.

ryaniskira, 2020-04-07:
I was fearing a hardware failure but RMA it is.

closed 2021-01-03 by RMerl