RMerl/asuswrt-merlin.ng #454: RT-AC86U loses packets and cannot access route

LJea, 2020-03-20:
Device: RT-AC86U
Version: I tried 384.12 to 384.15

When static routing is enabled, if a large amount of traffic (400 Mbps*2 downlink) is generated, packets may be lost and route cannot be accessed
The problem is solved after the static route is closed.

I suspect it has something to do with Dual WAN. The mode I use is Load Balance, and it works.
No related logging at debug level.

I'm not sure if this is a bug in Asus

LJea, 2020-03-23:
Turning off static routing still has this problem, but the chance of triggering is greatly reduced by 1/30. If this problem occurs, it will cause 5 to 15+ seconds of lan port packet loss.

I used multiple devices for testing and turned off the firewall, but this issue still exists

Syst3mSh0ck, 2020-03-28:
It could be an interface specific issue or perhaps packets are entering a policing queue somewhere. I would run a line test on both circuits/services and an iPerf tests if possible. Maybe something like an ICMP probe in PRTG to monitor both lines as well.

closed 2021-01-03 by RMerl